robotology/yarp #929: Ros topic interoperation instability when using UDP protocol

aerydna, 2016-10-17:
Disconnecting and re-connecting a yarp port that is publishing in a ros topic can cause the program to `exit()`.
I presume this to happens if the used protocol is the UDP because of the following:
If you publish on some topic, with Rviz you can subscribe to it, disconnect, re-connect and switch to UDP protocol by checking the "Unreliable" checkbox.
If you disconnect and re-connect while the checkbox is on somehow the yarp application will execute `carriers/xmlrpc_carrier/xmlrpc/XmlRpcValue::binaryFromXml()` that calls an `exit()` right after logging `"binaryFromXml disabled until license of base64.h determined\n"`.

drdanz, 2016-10-19:
@aerydna Closing, since I just merged #930 that *should* fix this issue. I cannot say for sure though, can you please test it again and reopen this issue if it is not fixed?

closed 2016-10-19 by drdanz

aerydna, 2016-10-19:
good! i'll test it as soon as possible..