robotology/yarp #936: yarpdataplayer: cannot load data dump with 1 entry

gsaponaro, 2016-10-24:
Hi @vtikha, I would like to use a data dump with exactly 1 entry of image type. I recorded it with yarpdatadumper (download link at the bottom), but when I try to load it in yarpdataplayer I get this error

```
There was a problem reading
data.log
please check its validity
```

I looked at the function ```Utilities::checkLogValidity(const char *filename)``` in ```utils.cpp``` which raises that error. While the file ```info.log``` passes the check (because it has 2 lines), ```data.log``` fails it (1 line).

One quick way to fix would be to change L229 from
```
            if (itr >= 1){
```
to
```
            if (itr >= 0){
```
(Or remove altogether the ```if```: L229 and L235.)

BTW, this is not urgent for me, I'm running the modification locally. Just wanted to make you aware of this corner case. :) Let me know if I should submit a PR.

PS: the data was acquired with ```yarpdatadumper --name /left --connect /icub/camcalib/left/out --type image --downsample 30``` and hitting ctrl+c after 1 second. [Here is the download link](http://users.isr.ist.utl.pt/~gsaponaro/temp/simple3.zip).

vtikha, 2016-10-26:
`yarpdataplayer` can now replay a one image dataset.

closed 2016-10-26 by vtikha

drdanz, 2016-10-26:
Fixed in 0307c0e7275420d19aebfcef615eefbf0f7ff349

drdanz, 2016-10-26:
And in a7f51e48c656a517a8ff08179f5eae9fde1c1dcb