robotology/yarp #937: os::Value issues while parsing/obtaining values

francesco-romano, 2016-10-25:
This issue is to summarise some of the common problems we find when using `os::Value`. 
Probably some of these can be fixed.

1) With the idea that a `Value` can be used to wrap a primitive type, there not exist a way to create a boolean value if not calling `fromString`, i.e.

2) When reading an int in a string, it is not possible to retrieve the value as a string.

3) there was a third issue which I do not remember now 😢 

The attached CMake and cpp code show the problem

```cmake
project(test_value)

find_package(YARP 2.3.66 REQUIRED)

include_directories(${YARP_INCLUDE_DIRS})

add_executable(test_value main.cpp)
target_link_libraries(test_value YARP::YARP_OS YARP::YARP_init)

```

cpp
```cpp
#include <yarp/os/Network.h>
#include <yarp/os/Value.h>

int main(int argc, char *argv[])
{
    std::string p = "31415";
    yarp::os::Value v;
    v.fromString(p.c_str());

    std::cout << "------- String / numeric ----------- " << std::endl;
    std::cout << v.toString() << std::endl;
    std::cout << "string:" << v.isString() << " " << v.asString() << std::endl;
    std::cout << "int:   " << v.isInt()    << " " << v.asInt() << std::endl;
    std::cout << "double:" << v.isDouble() << " " << v.asDouble() << std::endl;
    
    std::cout << "------- Bool ----------- " << std::endl;
    yarp::os::Value vBool1(true);
    std::cout << "Bool in constructor " << vBool1.isBool() << " " << vBool1.asBool() << std::endl;
    yarp::os::Value vBool2("true");
    std::cout << "Bool in constructor (str) " << vBool2.isBool() << " " << vBool2.asBool() << std::endl;
    yarp::os::Value vBool3; vBool3.fromString("true");
    std::cout << "Bool in fromString " << vBool3.isBool() << " " << vBool3.asBool() << std::endl;
}

```

Output: 
```sh
------- String / numeric ----------- 
31415
string:0 
int:   1 31415
double:0 31415
------- Bool ----------- 
Bool in constructor 0 1
Bool in constructor (str) 0 0
Bool in fromString 1 1
```

cc @drdanz @lornat75

randaz81, 2016-10-25:
I suppose that you'd like to have an output to similar to this (?):
```
------- String / numeric ----------- 
31415
string:0 31415 <---
int:   1 31415
double:0 31415
------- Bool ----------- 
Bool in constructor 1 1 <----
Bool in constructor (str) 0 0
Bool in fromString 1 1
```
I wonder also if a value containing a bool is converted to a string "true" or "false" when using asString(),  it might be helpful in some occasions.

francesco-romano, 2016-10-26:
@randaz81 yes.
Probably also in the second bool example I'd expect `Bool in constructor (str) 1 1`

drdanz, 2016-10-26:
The patch to implement that is trivial, the problem here is that there are tests that fail (on purpose) in this case.
I'm in favour to change this behaviour, I cannot really understand the reason why this is broken in the first place. Nonetheless, changing this is a change of behaviour and therefore risky, we have no idea where this is used and no way to find out if changing this will have some secondary effect.

pattacini, 2016-10-26:
- I'm in favor of fixing those things that are clearly bugs, no matter whether they'll have impact on the user code, because they're bugs after all.
- Instead, I really wouldn't change the current behavior to implement desired improvements, because of here we have a huge risk as @drdanz was saying.

pattacini, 2016-10-26:
Just to be more clear 😉 

A bug to be fixed is:
```cpp
yarp::os::Value v(true);
v.isBool()   \\ this must be true
```

A borderline behavior is instead:
```cpp
yarp::os::Value v("true");
v.isBool()   \\ this should remain a string ?
```

I'm afraid there's plenty of code where command line options are specified in terms of switches that are handled with `"true"` / `"false"` input values meant to be `string` rather than `bool`.

randaz81, 2016-10-26:
I'm also thinking that:

yarp::os::Value v("true");
v.isBool();   // this should remain a string

should remain a string...

drdanz, 2016-10-26:
Adding this:

```c++
    std::cout << "                   \t\t"       << "isBool"          << "\t"   << "asBool"          << "\t"
                                                 << "isInt"           << "\t"   << "asInt"           << "\t"
                                                 << "isDoub"          << "\t"   << "asDoub"          << "\t"
                                                 << "isStr"           << "\t"   << "asStr"           << "\t"     << "toStr"                   << std::endl;

    std::cout << "------- Bool ----------- " << std::endl;
    yarp::os::Value vBool1(true);
    yarp::os::Value vBool2("true");
    yarp::os::Value vBool3; vBool3.fromString("true");
    std::cout << "Bool in constructor\t\t"       << vBool1.isBool()   << "\t"   << vBool1.asBool()   << "\t"
                                                 << vBool1.isInt()    << "\t"   << vBool1.asInt()    << "\t"
                                                 << vBool1.isDouble() << "\t"   << vBool1.asDouble() << "\t"
                                                 << vBool1.isString() << "\t\"" << vBool1.asString() << "\"\t\"" << vBool1.toString() << "\"" << std::endl;
    std::cout << "Bool in constructor (str)\t"   << vBool2.isBool()   << "\t"   << vBool2.asBool()   << "\t"
                                                 << vBool2.isInt()    << "\t"   << vBool2.asInt()    << "\t"
                                                 << vBool2.isDouble() << "\t"   << vBool2.asDouble() << "\t"
                                                 << vBool2.isString() << "\t\"" << vBool2.asString() << "\"\t\"" << vBool2.toString() << "\"" << std::endl;
    std::cout << "Bool in fromString\t\t"        << vBool3.isBool()   << "\t"   << vBool3.asBool()   << "\t"
                                                 << vBool3.isInt()    << "\t"   << vBool3.asInt()    << "\t"
                                                 << vBool3.isDouble() << "\t"   << vBool3.asDouble() << "\t"
                                                 << vBool3.isString() << "\t\"" << vBool3.asString() << "\"\t\"" << vBool3.toString() << "\"" << std::endl;

    std::cout << "------- Int ------------ " << std::endl;
    yarp::os::Value vInt1(1);
    yarp::os::Value vInt2("1");
    yarp::os::Value vInt3; vInt3.fromString("1");
    std::cout << "Int in constructor\t\t"        << vInt1.isBool()    << "\t"   << vInt1.asBool()    << "\t"
                                                 << vInt1.isInt()     << "\t"   << vInt1.asInt()     << "\t"
                                                 << vInt1.isDouble()  << "\t"   << vInt1.asDouble()  << "\t"
                                                 << vInt1.isString()  << "\t\"" << vInt1.asString()  << "\"\t\"" << vInt1.toString()  << "\"" << std::endl;
    std::cout << "Int in constructor (str)\t"    << vInt2.isBool()    << "\t"   << vInt2.asBool()    << "\t"
                                                 << vInt2.isInt()     << "\t"   << vInt2.asInt()     << "\t"
                                                 << vInt2.isDouble()  << "\t"   << vInt2.asDouble()  << "\t"
                                                 << vInt2.isString()  << "\t\"" << vInt2.asString()  << "\"\t\"" << vInt2.toString()  << "\"" << std::endl;
    std::cout << "Int in fromString\t\t"         << vInt3.isBool()    << "\t"   << vInt3.asBool()    << "\t"
                                                 << vInt3.isInt()     << "\t"   << vInt3.asInt()     << "\t"
                                                 << vInt3.isDouble()  << "\t"   << vInt3.asDouble()  << "\t"
                                                 << vInt3.isString()  << "\t\"" << vInt3.asString()  << "\"\t\"" << vInt3.toString()  << "\"" << std::endl;

    std::cout << "------- Double --------- " << std::endl;
    yarp::os::Value vDoub1(1.);
    yarp::os::Value vDoub2("1.");
    yarp::os::Value vDoub3; vDoub3.fromString("1.");
    std::cout << "Double in constructor\t\t"     << vDoub1.isBool()   << "\t"   << vDoub1.asBool()   << "\t"
                                                 << vDoub1.isInt()    << "\t"   << vDoub1.asInt()    << "\t"
                                                 << vDoub1.isDouble() << "\t"   << vDoub1.asDouble() << "\t"
                                                 << vDoub1.isString() << "\t\"" << vDoub1.asString() << "\"\t\"" << vDoub1.toString()  << "\"" << std::endl;
    std::cout << "Double in constructor (str)\t" << vDoub2.isBool()   << "\t"   << vDoub2.asBool()   << "\t"
                                                 << vDoub2.isInt()    << "\t"   << vDoub2.asInt()    << "\t"
                                                 << vDoub2.isDouble() << "\t"   << vDoub2.asDouble() << "\t"
                                                 << vDoub2.isString() << "\t\"" << vDoub2.asString() << "\"\t\"" << vDoub2.toString()  << "\"" << std::endl;
    std::cout << "Double in fromString\t\t"      << vDoub3.isBool()   << "\t"   << vDoub3.asBool()   << "\t"
                                                 << vDoub3.isInt()    << "\t"   << vDoub3.asInt()    << "\t"
                                                 << vDoub3.isDouble() << "\t"   << vDoub3.asDouble() << "\t"
                                                 << vDoub3.isString() << "\t\"" << vDoub3.asString() << "\"\t\"" << vDoub3.toString()  << "\"" << std::endl;
```

Output:
```
[... cut ...]
Int in constructor (str)        0       0       0       0       0       0       1       "1"     "1"
Int in fromString               0       1       1       1       0       1       0       ""      "1"
------- Double --------- 
Double in constructor           0       0       0       1       1       1       0       ""      "1.0"
Double in constructor (str)     0       0       0       0       0       0       1       "1."    "1."
Double in fromString            0       0       0       1       1       1       0       ""      "1.0"
```

drdanz, 2016-10-26:
So, from what I understand:
* The ``Value(const ConstString &str)`` constructor always create a string.
* The ``fromString(const char *str)`` method is different, it "interprets" the string and create something depending on the content.

A few considerations about the API:
* There are static methods like `static Value* makeInt(int x)` for all the types except for bool, I think it makes sense to add a `makeBool(bool x)` method.
* These methods return a `Value*`. Ownership of the returned pointer is not clear from the documentation. I think returning a pointer from this kind of method is really ugly, I'd rather return a `Value`, I don't think we support any compiler that does not have RVO[¹](https://blog.knatten.org/2011/08/26/dont-be-afraid-of-returning-by-value-know-the-return-value-optimization/).
* A `Value::Value(bool)` method is missing, `Value(false)` uses the `int` constructor

pattacini, 2016-10-27:
A somewhat related issue we should tackle sooner or later is #712.

drdanz, 2017-01-12:
Added some related unit tests, see 629b1dfa749555bfaee05d3cd63c15a01739c65d