robotology/yarp #939: Fail to bind yarp to python

kkjh0723, 2016-10-27:
I tried python binding, following (http://wiki.icub.org/wiki/Python_bindings),
and get the result as follows,

    [100%] Built target _yarp
    Install the project...
    -- Install configuration: ""
    -- Installing: /usr/local/lib/python2.7/site-packages/yarp.py
    -- Installing: /usr/local/lib/python2.7/site-packages/_yarp.so
    -- Removed runtime path from "/usr/local/lib/python2.7/site-packages/_yarp.so"

I though the binding is completed.
Then, I tried import `yarp` in python but it doesn't work.

I also tried to add '/usr/local/lib/python2.7/site-packages/' to `LD_LIBRARY_PATH` but it seems not working.
I'm not sure what is the reason. There might be some steps I need to do after it?
yarp is working correctly in C++ though.

Currently, I'm using anaconda and it might be the reason...
Is there any other way if I use anaconda?

traversaro, 2016-10-27:
I think that the point is that `/usr/local/lib/python2.7/site-packages/` is not in `PYTHONPATH` for the Anaconda python, but is instead in `PYTHONPATH` for the system Python. 
Given that this installation directory is obtained by Python itself (see https://github.com/robotology/yarp/blob/master/bindings/CMakeLists.txt#L223) I am afraid you compiled the YARP Python bindings with system Python. If you want to use YARP Python bindings with Anaconda, you should ensure that the YARP build system find the Python you are interested in (the Anaconda one) while configuring. 
I think that this can be done by passing to CMake directly the expected results of the https://github.com/Kitware/CMake/blob/master/Modules/FindPythonInterp.cmake and https://github.com/Kitware/CMake/blob/master/Modules/FindPythonLibs.cmake modules.

kkjh0723, 2016-10-28:
Thanks very much @traversaro . That was the problem.
I change the install path (`CMAKE_INSTALL_PREFIX`) to anaconda folder using ccmake.
And it solves the problem.

I close this issue.

closed 2016-10-28 by kkjh0723