robotology/yarp #940: ROS to Yarp - Sending std_msgs INT8

vicentepedro, 2016-10-31:
Hi,

I was trying to receive an INT8 message from ROS using a yarp topic subcriber, however, I couldn't read the correct value on the yarp side.

If I use an Int16 message, it works fine.

drdanz, 2016-11-02:
I cannot reproduce this, sorry... It works for me...
Can you give me a minimal working example where this bug shows?

vicentepedro, 2016-11-02:
Hi, @drdanz 

I did some more tests in my module after your reply.

Actually, the received value is correct, but a cast (see below) is needed to print the value...
```
Subscriber<Int8> command_sub;
Int8 *commandreceived = new Int8();
commandreceived = command_sub.read();
cout << "Received command" << static_cast<unsigned> (commandreceived->data) << endl;
```
 I was expecting a different behaviour, but I realized that _data_ is a char.

drdanz, 2016-11-03:
Hi @vicentepedro, `char` is printed by `cout` as the character corresponding, not as a number.
on the other hand `unsigned char` is printed as a number.
The problem was probably that you were receiving some non printable characters, and therefore you could not see it printed.

Closing this issue, please reopen if you still have problems with this...

closed 2016-11-03 by drdanz

plinioMoreno, 2016-11-03:
Just a quick comment, the Int8 should be easily printed out as an integer, without having to do the cast because of its type of storage (char).