robotology/yarp #941: yarpdataplayer - strict option not resetting

arrenglover, 2016-11-03:
Currently when a dataset is loaded it is possible to select the "strict" protocol in the option menu. If the dataset is then replaced in the same window, the "strict" option remains checked. However, from my testing, the yarpdataplayer is no longer actually correctly using a strict protocol.

The current work-around is to un-check "strict" from the option menu and then re-check it to enable. Perhaps the behaviour should be the same as the similar option for "repeat" which is disabled (and correctly shown to be in the option menu) when the dataset is changed.

vtikha, 2016-11-03:
@arrenglover thanks for reporting. I have added the missing function as per your request.

arrenglover, 2016-11-03:
@vtikha confirmed that it is working as expected. thanks.

commit 9f04b34c (2016-11-03): [yarpdataplayer] resetting sendstrict upon new dataset

fixes #941