robotology/yarp #945: YARP with ENABLE_yarpcar_xmlrpc_carrier set to ON fails to compile in Visual Studio 2015

traversaro, 2016-11-04:
~~~
C:\Users\traversaro\Code\codyco-superbuild\external\YARP\extern\xmlrpcpp\xmlrpcpp\src\XmlRpcValue.cpp(433): error C2039: 'back_insert_iterator': non ▒ un membro di 'std' [C:\Users\traversaro\Code\codyco-superbuild\build\external\YARP\extern\xmlrpcpp\YARP_priv_xmlrpcpp.vcxproj] [C:\Users\traversaro\Code\codyco-superbuild\build\YARP.vcxproj]
  C:\Users\traversaro\Code\codyco-superbuild\external\YARP\extern\xmlrpcpp\xmlrpcpp\src\XmlRpcValue.cpp(433): error C2065: 'back_insert_iterator': identificatore non dichiarato [C:\Users\traversaro\Code\codyco-superbuild\build\external\YARP\extern\xmlrpcpp\YARP_priv_xmlrpcpp.vcxproj] [C:\Users\traversaro\Code\codyco-superbuild\build\YARP.vcxproj]
  C:\Users\traversaro\Code\codyco-superbuild\external\YARP\extern\xmlrpcpp\xmlrpcpp\src\XmlRpcValue.cpp(433): error C2275: 'YarpXmlRpc::XmlRpcValue::BinaryData': tipo non valido come espressione [C:\Users\traversaro\Code\codyco-superbuild\build\external\YARP\extern\xmlrpcpp\YARP_priv_xmlrpcpp.vcxproj] [C:\Users\traversaro\Code\codyco-superbuild\build\YARP.vcxproj]
  C:\Users\traversaro\Code\codyco-superbuild\external\YARP\extern\xmlrpcpp\xmlrpcpp\src\XmlRpcValue.cpp(433): error C2065: 'ins': identificatore non dichiarato [C:\Users\traversaro\Code\codyco-superbuild\build\external\YARP\extern\xmlrpcpp\YARP_priv_xmlrpcpp.vcxproj] [C:\Users\traversaro\Code\codyco-superbuild\build\YARP.vcxproj]
  C:\Users\traversaro\Code\codyco-superbuild\external\YARP\extern\xmlrpcpp\xmlrpcpp\src\XmlRpcValue.cpp(433): error C2039: 'back_inserter': non ▒ un membro di 'std' [C:\Users\traversaro\Code\codyco-superbuild\build\external\YARP\extern\xmlrpcpp\YARP_priv_xmlrpcpp.vcxproj] [C:\Users\traversaro\Code\codyco-superbuild\build\YARP.vcxproj]
  C:\Users\traversaro\Code\codyco-superbuild\external\YARP\extern\xmlrpcpp\xmlrpcpp\src\XmlRpcValue.cpp(433): error C3861: 'back_inserter': identificatore non trovato [C:\Users\traversaro\Code\codyco-superbuild\build\external\YARP\extern\xmlrpcpp\YARP_priv_xmlrpcpp.vcxproj] [C:\Users\traversaro\Code\codyco-superbuild\build\YARP.vcxproj]
  C:\Users\traversaro\Code\codyco-superbuild\external\YARP\extern\xmlrpcpp\xmlrpcpp\src\XmlRpcValue.cpp(434): error C2065: 'ins': identificatore non dichiarato [C:\Users\traversaro\Code\codyco-superbuild\build\external\YARP\extern\xmlrpcpp\YARP_priv_xmlrpcpp.vcxproj] [C:\Users\traversaro\Code\codyco-superbuild\build\YARP.vcxproj]
  C:\Users\traversaro\Code\codyco-superbuild\external\YARP\extern\xmlrpcpp\xmlrpcpp\src\XmlRpcValue.cpp(433): error C2039: 'back_insert_iterator': non ▒ un membro di 'std' [C:\Users\traversaro\Code\codyco-superbuild\build\external\YARP\extern\xmlrpcpp\YARP_priv_xmlrpcpp.vcxproj] [C:\Users\traversaro\Code\codyco-superbuild\build\YARP.vcxproj]
  C:\Users\traversaro\Code\codyco-superbuild\external\YARP\extern\xmlrpcpp\xmlrpcpp\src\XmlRpcValue.cpp(433): error C2065: 'back_insert_iterator': identificatore non dichiarato [C:\Users\traversaro\Code\codyco-superbuild\build\external\YARP\extern\xmlrpcpp\YARP_priv_xmlrpcpp.vcxproj] [C:\Users\traversaro\Code\codyco-superbuild\build\YARP.vcxproj]
  C:\Users\traversaro\Code\codyco-superbuild\external\YARP\extern\xmlrpcpp\xmlrpcpp\src\XmlRpcValue.cpp(433): error C2275: 'YarpXmlRpc::XmlRpcValue::BinaryData': tipo non valido come espressione [C:\Users\traversaro\Code\codyco-superbuild\build\external\YARP\extern\xmlrpcpp\YARP_priv_xmlrpcpp.vcxproj] [C:\Users\traversaro\Code\codyco-superbuild\build\YARP.vcxproj]
  C:\Users\traversaro\Code\codyco-superbuild\external\YARP\extern\xmlrpcpp\xmlrpcpp\src\XmlRpcValue.cpp(433): error C2065: 'ins': identificatore non dichiarato [C:\Users\traversaro\Code\codyco-superbuild\build\external\YARP\extern\xmlrpcpp\YARP_priv_xmlrpcpp.vcxproj] [C:\Users\traversaro\Code\codyco-superbuild\build\YARP.vcxproj]
  C:\Users\traversaro\Code\codyco-superbuild\external\YARP\extern\xmlrpcpp\xmlrpcpp\src\XmlRpcValue.cpp(433): error C2039: 'back_inserter': non ▒ un membro di 'std' [C:\Users\traversaro\Code\codyco-superbuild\build\external\YARP\extern\xmlrpcpp\YARP_priv_xmlrpcpp.vcxproj] [C:\Users\traversaro\Code\codyco-superbuild\build\YARP.vcxproj]
  C:\Users\traversaro\Code\codyco-superbuild\external\YARP\extern\xmlrpcpp\xmlrpcpp\src\XmlRpcValue.cpp(433): error C3861: 'back_inserter': identificatore non trovato [C:\Users\traversaro\Code\codyco-superbuild\build\external\YARP\extern\xmlrpcpp\YARP_priv_xmlrpcpp.vcxproj] [C:\Users\traversaro\Code\codyco-superbuild\build\YARP.vcxproj]
  C:\Users\traversaro\Code\codyco-superbuild\external\YARP\extern\xmlrpcpp\xmlrpcpp\src\XmlRpcValue.cpp(434): error C2065: 'ins': identificatore non dichiarato [C:\Users\traversaro\Code\codyco-superbuild\build\external\YARP\extern\xmlrpcpp\YARP_priv_xmlrpcpp.vcxproj] [C:\Users\traversaro\Code\codyco-superbuild\build\YARP.vcxproj]
~~~

Sorry for the italian error messages. : (

traversaro, 2016-11-04:
This is fork that may have the necessary patches to compile correctly in VC2015 : https://github.com/kenjinote/xmlrpc .

drdanz, 2016-11-08:
Since in devel xmlrpc is now always compiled, I consider this as a blocker for the next release.

drdanz, 2016-11-14:
Fixed in 2e814e8

commit 2e814e8c (2016-11-14): xmlrpcpp: Add missing include

Fixes #945
closed 2016-11-14 by drdanz