robotology/yarp #948: If YARP is compiled without ACE, loading plugin silently fails

traversaro, 2016-11-08:
See https://github.com/robotology/QA/issues/148 .

It should print a clear error message on failure, to help users debug the problem. 

@drdanz @pattacini This issue is a good candidate for newcomers at IFI day, do we have a label for this?

drdanz, 2016-11-08:
This is already fixed in devel, see robotology/yarp@d117fffc0392a5cb13a7b0b9352b5825a0aa4785.
I don't think it is worth fixing this in master... @traversaro what do you think?

@traversaro We do have a label for IFI days ("Priority: IFI Day"), but it does not say anything about the "complexity" to fix a issue, maybe we could add some "Complexity: Trivial/Low/Medium/Hard" labels?

traversaro, 2016-11-08:
@drdanz 
I totally forgot that! :D 
I think we can close this issue and find some other useful issues for newcomers.

closed 2016-11-08 by traversaro