robotology/yarp #956: BottleImpl.h:114:5: warning: all paths through this function will call itself

traversaro, 2016-11-14:
Interesting warning found by clang on macOS: 
~~~
In file included from /Users/traversaro/src/codyco-superbuild/external/YARP/src/libYARP_OS/src/Value.cpp:12:
/Users/traversaro/src/codyco-superbuild/external/YARP/src/libYARP_OS/include/yarp/os/impl/BottleImpl.h:114:5: warning: all paths through this function will call itself [-Winfinite-recursion]
    {
    ^
1 warning generated.
~~~
Probably related to : https://github.com/robotology/yarp/pull/938

traversaro, 2016-11-17:
In particular: https://github.com/robotology/yarp/pull/938/files#diff-00753941a4fed40b6cecb12cea19b407R115 .

drdanz, 2016-11-17:
This is the kind of things that can happen when the unit tests for the main classes are not complete! :stuck_out_tongue_closed_eyes:

closed 2016-11-18 by drdanz