robotology/yarp #972: Error reading from a serial port

vickyvou, 2016-11-23:
Dear all,
i am trying to read some values sent from an arduino through a serial port in yarp. In my files i have:
```
   #include <yarp/dev/SerialInterfaces.h>
   #include <yarp/dev/PolyDriver.h>
   PolyDriver driver;
   ISerialDevice *iSerial;

    Property prop;
    prop.put("device", "serialport");
    prop.put("comport", "/dev/tty.usbmodemFD131");
    prop.put("baudrate", 9600);
    prop.put("paritymode", "NONE");
    prop.put("databits", 8);
    prop.put("stopbits", 2);
    prop.put("rcvenb", 1);
    prop.put("verbose", 0);

driver.open(prop);
    if(!driver.isValid()){
        fprintf(stderr, "Error opening PolyDriver check parameters\n");
    }
driver.view(iSerial);
    if(!iSerial) {
        fprintf(stderr, "Error opening serial driver. Device not available\n");
    }
```

as proposed by another thread. 
My dev list is the following:
```
ere are devices listed for your system:
Device "test_grabber", C++ class yarp::dev::TestFrameGrabber, wrapped by "grabber"
Device "test_motor", C++ class yarp::dev::TestMotor, wrapped by "controlboard"
Device "remote_grabber", C++ class yarp::dev::RemoteFrameGrabber, wrapped by "grabber"
Device "grabber", C++ class yarp::dev::ServerFrameGrabber, is a network wrapper.
Device "sound_grabber", C++ class yarp::dev::ServerSoundGrabber, is a network wrapper.
Device "pipe", C++ class yarp::dev::DevicePipe, has no network wrapper
Device "group", C++ class yarp::dev::DeviceGroup, has no network wrapper
Device "inertial", C++ class yarp::dev::ServerInertial, is a network wrapper.
Device "controlboardwrapper2", C++ class yarp::dev::ControlBoardWrapper2, is a network wrapper.
Device "remote_controlboard", C++ class yarp::dev::RemoteControlBoard, wrapped by "controlboardwrapper2"
Device "batteryClient", C++ class batteryClient, has no network wrapper
Device "Rangefinder2DClient", C++ class Rangefinder2DClient, has no network wrapper
Device "analogsensorclient", C++ class AnalogSensorClient, has no network wrapper
Device "analogServer", C++ class yarp::dev::AnalogWrapper, is a network wrapper.
Device "virtualAnalogServer", C++ class yarp::dev::VirtualAnalogWrapper, is a network wrapper.
Device "batteryWrapper", C++ class yarp::dev::BatteryWrapper, is a network wrapper.
Device "Rangefinder2DWrapper", C++ class yarp::dev::Rangefinder2Dwrapper, is a network wrapper.
Device "clientcontrolboard", C++ class yarp::dev::ClientControlBoard, wrapped by "controlboard"
Device "controlboard", C++ class yarp::dev::ServerControlBoard, is a network wrapper.
Device "RGBDSensorWrapper", C++ class yarp::dev::RGBDSensorWrapper, is a network wrapper.
Device "RGBDSensorClient", C++ class yarp::dev::RGBDSensorClient, is a network wrapper.
Device "serial", available on request (found in /Users/VickySpecs/robots/yarp/build/lib/yarp/yarp_serial.so library)is a network wrapper..
Device "serialport", available on request (found in /Users/VickySpecs/robots/yarp/build/lib/yarp/yarp_serial.so library), wrapped by "serial".
Device "cartesiancontrollerclient", available on request (found in /Users/VickySpecs/robots/icub-main/build/lib/iCub/cartesiancontrollerclient.so library)is a network wrapper..
Device "cartesiancontrollerserver", available on request (found in /Users/VickySpecs/robots/icub-main/build/lib/iCub/cartesiancontrollerserver.so library), wrapped by "cartesiancontrollerclient".
Device "debugInterfaceClient", available on request (found in /Users/VickySpecs/robots/icub-main/build/lib/iCub/debugInterfaceClient.so library).
Device "gazecontrollerclient", available on request (found in /Users/VickySpecs/robots/icub-main/build/lib/iCub/gazecontrollerclient.so library)is a network wrapper..
```
However, when i try to run the module it gets stuck at the driver, and i get the following errors:
```
[ERROR]Invalid communications port in /dev/tty.usbmodemFD131 
yarpdev: ***ERROR*** driver <serialport> was found but could not open
Error opening PolyDriver check parameters
Error opening serial driver. Device not available
```
Any ideas on what could be the problem? thanks!

randaz81, 2016-11-23:
Hi! 

1) Does your user have correct permission to access /dev/tty.usbmodemFD131? if root is owner of the device, you can set a rule or change the owner with `chown `command
2) This is a more complete list of all parameters used on our robot. I do not remember which of them are mandatory, but it's a good idea to set all of them and do not use default values.
```
verbose 4
comport /dev/ttyUSB0
baudrate 115200
xonlim 0
xofflim 0
readmincharacters 0
readtimeoutmsec 2
parityenb 0
paritymode none
ctsenb 0
rtsenb 0
xinenb 0
xoutenb 0
modem 0
rcvenb 0
dsrenb 0
dtrdisable 1
databits 8
stopbits 1
line_terminator_char1 10
line_terminator_char2 10
```

randaz81, 2016-11-23:
PS: you can also check `yarp\src\devices\rpLidar` for a working example of a seriaport device.

vickyvou, 2016-11-23:
Thank you for the quick reply!
I checked the permissions, i changed them, i ran my module and i still got the same problems. But in general i can read from that port the information sent from the terminal (in case it helps, it is a mac)
I will check the example and let you know! 

I cannot understand what might be the reason because the code added comes from an example previously mentioned in another thread and also, we had implemented (code is extremely similar) previously (during EFAA) a similar module and it was working. Is there anything else i can check that might be helpful?

vickyvou, 2016-11-24:
Update:
i was playing a bit with the different permissions, so when the permissions are "root" i get the following error:
```
[ERROR]Invalid communications port in /dev/tty.usbmodemFD131 
yarpdev: ***ERROR*** driver <serialport> was found but could not open
Error opening PolyDriver check parameters
Error opening serial driver. Device not available
```

if i change the permissions to my current user and i run the module i get the exact same error. In both cases, the error appears when i try to close the application in the console, if not it is stuck there. 

Now, if i run the module with sudo i get the following error:
```
[ERROR]Cannot find "serialport" plugin (not built in, and no .ini file found for it)
[ERROR]Check that YARP_DATA_DIRS leads to at least one directory with plugins/serialport.ini or share/yarp/plugins/serialport.ini in it
yarpdev: ***ERROR*** could not find device <serialport>
Error opening PolyDriver check parameters
Error opening serial driver. Device not available
```

So i am not sure i can understand what is the problem...

randaz81, 2016-11-24:
You cannot run your module with sudo, root has different environment variables. In particular if YARP_DATA_DIRS is missing, the module cannot load the serialport plugin.

randaz81, 2016-11-24:
> if i change the permissions to my current user and i run the module i get the exact same 
 error

I'm not sure if I understood correctly. Can you please tell me the output of `ls /dev/ -la`

vickyvou, 2016-11-24:
Oh yes, true! Sorry, i have am quite desperate to make this work and didn't think of that at all!
the output is:

```
ls -l /dev/tty.*
crw-rw-rw-  1 root        wheel   17,   0 Nov 23 14:52 /dev/tty.Bluetooth-Incoming-Port
crw-rw-rw-  1 VickySpecs  wheel   17,  42 Nov 24 14:05 /dev/tty.usbmodemFD131
```

randaz81, 2016-11-25:
sorry @vickyvou, I can't find an explanation for your problem. Permissions seem ok. From the source code I noticed that this error is printed even before evaluating the serial port configuration: the name is the only parameter checked at that point. So, without having physical access to your system and launch a debugger on your machine I cannot help you. :-(
I suggest you to go on by making attempts as follows:
- check the same arduino device on a different MACOS machine 
- check the same arduino device on a different OS (linux, Win32 ecc)
- check a different serial device on the same MACOS machine
- create a symbolic link to /dev/tty.usbmodemFD131 using a name that does not contain the . character
- ... (still thinking) ....

randaz81, 2016-11-25:
I just tested the seriaport driver on MAC with @vtikha and it works...

vickyvou, 2016-11-25:
Hey, thank you very much for your reply. 

So, i tried many different things, i could read without a problem from the serial port, the arduino works in both my mac (also using the arduino software or terminal) and also in a windows machine. I also checked the same arduino with different usb ports. I also made sure that no other sw is accessing that port, but i might try the symbolic link, just to see. I tried to use the windows machine with the same code but i got compilation errors with yarp and the corresponding librabry (that is another issue i will submit later) and didn't work.

 So i gave up and i used c++ code without yarp to read from the port, which works quite well (on windows though) and then i just forward the info to a port to my main module. A quick but not elegant solution. It is a pity because it would be nice because it definitely makes things easier. It could be that it is something with my computer, i have no idea. 

Thank you very much for your time, in any case!

drdanz, 2016-11-28:
@vickyvou `sudo` does not pass all the environment variables to the executed command, therefore when you run your executable with sudo `YARP_DATA_DIRS` is not set, and YARP cannot find the `serial` device.

Can you try `sudo env YARP_DATA_DIRS=$YARP_DATA_DIRS ./test_serial` and let us know if it works?

drdanz, 2016-11-28:
@vickyvou I just opened a pull request (#976) that prints a better error when the device cannot be opened. This should help debugging this issue. I will merge it as soon as I'm sure that it builds everywhere.

closed 2016-12-06 by drdanz