robotology/yarp #975: yarpdev silently fails if a .ini device manifest has a wrong library parameter

traversaro, 2016-11-25:
It the .ini plugin manifest has a wrong library name, `yarpdev --list` silently skips that library without any warning. Having a little warning in this case would drastically help when developing new yarp devices.

bhigy, 2016-12-01:
I added some code to output a warning but then I get a lot of them (see output below). Does that sound plausible?

```
bhigy@iiticublap019:~/dev/yarp/build$ yarpdev --list
Here are devices listed for your system:
[WARNING]Wrong library name for plugin SerialServoBoard
[WARNING]Wrong library name for plugin cuda
[WARNING]Wrong library name for plugin dynamixelAX12Ftdi
[WARNING]Wrong library name for plugin fakeIMU
[WARNING]Wrong library name for plugin fakeLaser
[WARNING]Wrong library name for plugin fakeMotionControl
[WARNING]Wrong library name for plugin fakebot
[WARNING]Wrong library name for plugin ffmpeg_grabber
[WARNING]Wrong library name for plugin ffmpeg_writer
[WARNING]Wrong library name for plugin imuBosch_BNO055
[WARNING]Wrong library name for plugin KinectDeviceLocal
[WARNING]Wrong library name for plugin KinectDeviceClient
[WARNING]Wrong library name for plugin laserHokuyo
[WARNING]Wrong library name for plugin microphone
[WARNING]Wrong library name for plugin nvidia
[WARNING]Wrong library name for plugin opencv_grabber
[WARNING]Wrong library name for plugin OpenNI2DeviceServer
[WARNING]Wrong library name for plugin OpenNI2DeviceClient
[WARNING]Wrong library name for plugin ovrheadset
[WARNING]Wrong library name for plugin rpLidar
[WARNING]Wrong library name for plugin serial
[WARNING]Wrong library name for plugin serialport
[WARNING]Wrong library name for plugin stage
[WARNING]Wrong library name for plugin wxsdl
Device "test_grabber", C++ class yarp::dev::TestFrameGrabber, wrapped by "grabber"
Device "test_motor", C++ class yarp::dev::TestMotor, wrapped by "controlboard"
Device "remote_grabber", C++ class yarp::dev::RemoteFrameGrabber, wrapped by "grabber"
Device "grabber", C++ class yarp::dev::ServerFrameGrabber, is a network wrapper.
Device "sound_grabber", C++ class yarp::dev::ServerSoundGrabber, is a network wrapper.
Device "pipe", C++ class yarp::dev::DevicePipe, has no network wrapper
Device "group", C++ class yarp::dev::DeviceGroup, has no network wrapper
Device "remote_controlboard", C++ class yarp::dev::RemoteControlBoard, wrapped by "controlboardwrapper2"
Device "inertial", C++ class yarp::dev::ServerInertial, is a network wrapper.
Device "controlboardwrapper2", C++ class yarp::dev::ControlBoardWrapper2, is a network wrapper.
Device "batteryClient", C++ class batteryClient, has no network wrapper
Device "Rangefinder2DClient", C++ class Rangefinder2DClient, has no network wrapper
Device "analogsensorclient", C++ class AnalogSensorClient, has no network wrapper
Device "analogServer", C++ class yarp::dev::AnalogWrapper, is a network wrapper.
Device "virtualAnalogServer", C++ class yarp::dev::VirtualAnalogWrapper, is a network wrapper.
Device "batteryWrapper", C++ class yarp::dev::BatteryWrapper, is a network wrapper.
Device "Rangefinder2DWrapper", C++ class yarp::dev::Rangefinder2Dwrapper, is a network wrapper.
Device "clientcontrolboard", C++ class yarp::dev::ClientControlBoard, wrapped by "controlboard"
Device "controlboard", C++ class yarp::dev::ServerControlBoard, is a network wrapper.
Device "portaudio", available on request (found in /home/bhigy/dev/yarp/build/lib/yarp/yarp_portaudio.so library), wrapped by "grabber".
Device "cartesiancontrollerclient", available on request (found in /home/bhigy/dev/icub-main/build/lib/iCub/cartesiancontrollerclient.so library), is a network wrapper.
Device "cartesiancontrollerserver", available on request (found in /home/bhigy/dev/icub-main/build/lib/iCub/cartesiancontrollerserver.so library), wrapped by "cartesiancontrollerclient".
Device "gazecontrollerclient", available on request (found in /home/bhigy/dev/icub-main/build/lib/iCub/gazecontrollerclient.so library), is a network wrapper.
Device "speech", available on request (found in /home/bhigy/lib/speech/svox_speech.so library).
```

traversaro, 2016-12-01:
Can you push the code to a branch on your fork so we can check it? Thanks!

bhigy, 2016-12-01:
Done

drdanz, 2016-12-01:
What does `yarp plugin` report in this case?

bhigy, 2016-12-02:
I tested the first ones and I get
```
Yes, this is a YARP plugin
    Cannot find or load shared library
```

So I guess it's working properly. :)

drdanz, 2016-12-02:
Fixed in e55501597ad9854bec95238181df170950be0c90. Thanks @bhigy.

closed 2016-12-02 by drdanz