robotology/yarp #980: Wrong YARP_PORT_PREFIX in yarpmanager

Tobias-Fischer, 2016-11-30:
Hi, currently the `YARP_PORT_PREFIX=;`, which breaks some applications (e.g. our robot didn't start using `yarprobotinterface` - not sure whether this is related or some other issue with our robot).
This is due to https://github.com/robotology/yarp/blob/master/src/libYARP_manager/src/manager.cpp#L292 which was introduced in https://github.com/robotology/yarp/commit/238bf639343860921b3bb68e394d2bcdd1f12c0a

apaikan, 2016-12-01:
`YARP_PORT_PREFIX=;` is the same as the previous (i.e., `YARP_PORT_PREFIX=`) in the yarpmanager (theoretically should be the same). 

@Tobias-Fischer  would you provide more information on "it breaks some applications"? 
- Dose the manager run your modules? 
- What is the output of `yarp name list` after launching the application/modules? 
- With which particular module  you faced this problem?

maxime-petit, 2016-12-01:
Hi,

- Yes the manager does run modules
- With the `;` symbol, for instance when running yarprobotinterface, the port for the analogServer is wrong: `yarp: Port ;/icub/left_hand/analog:o/rpc:i active at tcp://10.0.2.2:10065` which is then elading to the yarpdev error `ERROR	AnalogServer Error initializing YARP ports` and `yarpdev: ***ERROR*** driver <analogServer> was found but could not open`
- When removing the `;` yarprobotinterface is starting nicely

./Max and @Tobias-Fischer

apaikan, 2016-12-01:
I guess your YARP is not updated on the machine (e.g. PC104) where you are running `yarprobotinterface`.  The https://github.com/robotology/yarp/commit/238bf639343860921b3bb68e394d2bcdd1f12c0a also modified `yarprun`.!!! 

Anyway, I just changed the yarp manager library to double check the port prefix and environment variables to avoid unnecessary "`;`".  
See pull request https://github.com/robotology/yarp/pull/982

@maxime-petit would you please update your YARP on PC104 and let me know?

Tobias-Fischer, 2016-12-20:
Hi @apaikan,
We got rid of this error on all of our Linux machines after updating, however our Windows machine still suffers from the ";" in front of the port names. We use the Windows machine mainly for `speechRecognizer`, but I guess the problem is independent of that. Could you please have a look whether it's working on your Windows machines?

Thanks,
Tobias

apaikan, 2017-01-16:
Hi
 @Tobias-Fischer Do you still getting ';' in front of the ports prefix in Windows? 
I have just checked it on my windows machine (devel branch) and everything look good. 
Would you check it again (e.g. with `EyesViewer-Localhost` from yarpmanager)

Tobias-Fischer, 2017-01-20:
Hi @apaikan,
I completely re-installed yarp on the Windows machine, and I think it's all good now. Thanks and sorry for the confusion.

Best, Tobi

closed 2017-01-20 by Tobias-Fischer