robotology/yarp #998: add to docs: extra comments may prevent thrift compilation

tanismar, 2016-12-05:
Hi there, 

On some systems, extra comments on the thrift pages might prevent this files from compiling. As it is not an intuitive thing, and tricky to debug (as usually comments are skipped on debugging), it should be advised in the thrift documentation to keep the comments for function documentation only. 

Example: 
In my case, I had a thrift file with many functions, so I made some comment separators for easier reading, as in the following: 

`    
    /*********************** FOOs ************************************/
    /**
     * Bla bla
     * @return true/false on success/failure
     */
    bool foo1(1: i32 x = 1);

    /*********************** FIIs ************************************/
    /**
     * Bli bli
     * @return true/false on success/failure to perform all actions
     */
    bool fii(1: i32 x, 2: i32 y = 0);
`

The separators had to be removed in order for the program to compile. 
/*********************** FXXs ************************************/ 

. 

Best, 
Tanis

pattacini, 2016-12-05:
Hi @tanismar 

What about creating a PR to fix this up? 😉 
Maybe at next IFI?

drdanz, 2016-12-06:
We could also consider fixing this in the thrift compiler...
It annoys me to know that if I add some comment the file will not compile.

drdanz, 2021-09-09:
This is a problem of the thrift parser which we just import but we do not want to modify, and therefore should eventually be fixed upstream.

Closing this.

closed 2021-09-09 by drdanz