robotology/yarp #1011: cannot compile yarp (cmake error)

serena-ivaldi, 2016-12-12:
Hello
I just downloaded the last version of yarp on a virgin Ubuntu 14 machine and I cannot compile it.

git log says:
commit 729eca738ed1b73bf4fb71c8e111892a87cdd654
Merge: 2e0e4e9 1e3b06c
Author: Daniele E. Domenichelli <daniele.domenichelli@gmail.com>
Date:   Mon Dec 12 10:40:50 2016 +0100

I tried to compile the project but I can't, and I am receiving the following error:

 CMake Error at cmake/YarpSystemCheck.cmake:131 (check_include_file_cxx):
   Unknown CMake command "check_include_file_cxx".
 Call Stack (most recent call first):
   CMakeLists.txt:50 (include)

By the way there is a flag YCM_DIR not found - and I do not see anything else in the cmake.
Any ideas?

Thanks in advance

traversaro, 2016-12-13:
This is strange, we have all our testing environment on Travis based on Ubuntu 14.04 (I assume you mean that by Ubuntu 14) and we did not catch any error. 

Could you provide the output of the following commands:
~~~
cmake --version 
gcc --version
~~~

And the content of the contents of the files `CMakeFiles/CMakeOutput.log` and `CMakeFiles/CMakeError.log` in your build? Thanks.

serena-ivaldi, 2016-12-13:
cmake version 2.8.12.2
gcc (Ubuntu 4.8.4-2ubuntu1~14.04.3) 4.8.4

[CMakeOutput_log.txt](https://github.com/robotology/yarp/files/649329/CMakeOutput_log.txt)
[CMakeError_log.txt](https://github.com/robotology/yarp/files/649331/CMakeError_log.txt)

traversaro, 2016-12-13:
Can you also attach the `CMakeCache.txt` file?

serena-ivaldi, 2016-12-13:
[CMakeCache.txt](https://github.com/robotology/yarp/files/649350/CMakeCache.txt)

drdanz, 2016-12-13:
@serena-ivaldi It should be fixed now, can you pull the latest changes and try again please?

closed 2016-12-13 by drdanz

serena-ivaldi, 2016-12-13:
yes now cmake is able to go on! thanks!

commit 0389d9c7 (2016-12-13): CMake: Add missing include

Fixes #1011