robotology/yarp #1024: Python binding not compiling

charleswilmot, 2016-12-14:
Hello,

I try to compile YARP v2.3.68 together with the Python bindings.

```
git clone https://github.com/robotology/yarp.git
mkdir build
cd build
ccmake .. #latest ccmake version

# set CREATE_GUIS to ON
# set CREATE_LIB_MATH to ON
# check: CREATE_SHARED_LIBRARY is set to ON
# set YARP_COMPILE_BINDINGS to ON
# note: YCM_DIR-NOTFOUND
# configure
# warnings COPY_ONLY --> for project developers
# set CREATE_PYTHON to ON
# configure
# set YARP_USE_PYTHON_VERSION to 3.4
# toggle advanced mode
# set CMAKE_INSTALL_PYTHONDIR to lib/python3.4/dist-package
# set PYTHON_EXECUTABLE to /usr/bin/python3
# set PYTHON_INCLUDE_DIR to /usr/include/python3.4m
# set PYTHON_LIBRARY to /usr/lib/x86_64-linux-gnu/libpython3.4m.so
# configure
# generate

make
```

The compilation works fine untill 99%, when it starts compiling the wrapper *yarpPYTHON_warp.cxx*.

```
[100%] Building CXX object bindings/CMakeFiles/_yarp_python.dir/__/lib/python/yarpPYTHON_wrap.cxx.o

[... 2 warnings here ...]

/yarp/build/lib/python/yarpPYTHON_wrap.cxx: In function ‘PyObject* _wrap_Contact_setNested(PyObject*, PyObject*)’:
/yarp/build/lib/python/yarpPYTHON_wrap.cxx:11228:13: error: ‘class yarp::os::Contact’ has no member named ‘setNested’
     (arg1)->setNested((yarp::os::NestedContact const &)*arg2);

/yarp/build/lib/python/yarpPYTHON_wrap.cxx: In function ‘PyObject* _wrap_Matrix_getGslMatrix__SWIG_0(PyObject*, PyObject*)’:
/yarp/build/lib/python/yarpPYTHON_wrap.cxx:59115:30: error: ‘class yarp::sig::Matrix’ has no member named ‘getGslMatrix’
     result = (void *)(arg1)->getGslMatrix();

/yarp/build/lib/python/yarpPYTHON_wrap.cxx: In function ‘PyObject* _wrap_Matrix_getGslMatrix__SWIG_1(PyObject*, PyObject*)’:
/yarp/build/lib/python/yarpPYTHON_wrap.cxx:59141:57: error: ‘const class yarp::sig::Matrix’ has no member named ‘getGslMatrix’
     result = (void *)((yarp::sig::Matrix const *)arg1)->getGslMatrix();

/yarp/build/lib/python/yarpPYTHON_wrap.cxx: In function ‘PyObject* _wrap_Vector_getGslVector__SWIG_0(PyObject*, PyObject*)’:
/yarp/build/lib/python/yarpPYTHON_wrap.cxx:60699:30: error: ‘class yarp::sig::Vector’ has no member named ‘getGslVector’
     result = (void *)(arg1)->getGslVector();

/yarp/build/lib/python/yarpPYTHON_wrap.cxx: In function ‘PyObject* _wrap_Vector_getGslVector__SWIG_1(PyObject*, PyObject*)’:
/yarp/build/lib/python/yarpPYTHON_wrap.cxx:60725:57: error: ‘const class yarp::sig::Vector’ has no member named ‘getGslVector’
     result = (void *)((yarp::sig::Vector const *)arg1)->getGslVector();

[... 1 more warning here ...]

/yarp/build/lib/python/yarpPYTHON_wrap.cxx: In function ‘PyObject* PyInit__yarp()’:
/yarp/build/lib/python/yarpPYTHON_wrap.cxx:135762:88: error: ‘YARP_FEATURE_NUMEBR_OF’ was not declared in this scope
   SWIG_Python_SetConstant(d, "YARP_FEATURE_NUMEBR_OF",SWIG_From_int(static_cast< int >(YARP_FEATURE_NUMEBR_OF)));

make[2]: *** [bindings/CMakeFiles/_yarp_python.dir/__/lib/python/yarpPYTHON_wrap.cxx.o] Error 1
make[1]: *** [bindings/CMakeFiles/_yarp_python.dir/all] Error 2
make: *** [all] Error 2
```

For example, the class [yarp::os::Contact](http://www.yarp.it/classyarp_1_1os_1_1Contact.html#a303a4ddcfeb12fca2c8a5a3a265460fb) has no member *setNested*, but has a member *setNestedContact*... Not sure if it helps.

Also note that YARP v2.3.66 is installed on my lab's network from the deb package (without bindings). (Conflicts have been reported, see issue [#658](https://github.com/robotology/yarp/issues/658))

My issue seems similar to [#749](https://github.com/robotology/yarp/issues/749)

traversaro, 2016-12-14:
The errors are due to the fact that SWIG is finding the `v2.3.66` headers installed by the deb package before the local include on the `v2.3.68`. This is indeed a problem in the SWIG build system (the `build` include directories should have an higher priority) but the obvious solution in you case (if it is possible) would be to uninstall the `v2.3.66` debian.

traversaro, 2016-12-14:
I did a quick test, and I am afraid that giving priority to the system include paths is a behavior of SWIG, and could be difficult to fix. 
Can you run `VERBOSE=1 make` and copy paste the line after the `Swig source` message, i.e. the one similar to the following one:
~~~
[ 99%] Swig source
cd /Users/traversaro/src/codyco-superbuild/build/external/YARP/bindings && /usr/local/Cellar/cmake/3.6.3/bin/cmake -E make_directory /Users/traversaro/src/codyco-superbuild/build/external/YARP/lib/python
cd /Users/traversaro/src/codyco-superbuild/build/external/YARP/bindings && /usr/local/bin/swig -python -Wall -module yarp -threads -outdir /Users/traversaro/src/codyco-superbuild/build/external/YARP/lib/python -c++ -I/Users/traversaro/src/codyco-superbuild/build/external/YARP/generated_include -I/Users/traversaro/src/codyco-superbuild/external/YARP/src/libYARP_OS/include -I/Users/traversaro/src/codyco-superbuild/external/YARP/src/libYARP_sig/include -I/Users/traversaro/src/codyco-superbuild/external/YARP/src/libYARP_eigen/include -I/Users/traversaro/src/codyco-superbuild/external/YARP/src/libYARP_gsl/include -I/Users/traversaro/src/codyco-superbuild/external/YARP/src/libYARP_dev/include -I/Users/traversaro/src/codyco-superbuild/external/YARP/src/libYARP_name/include -I/Users/traversaro/src/codyco-superbuild/external/YARP/src/libYARP_manager/include -I/Users/traversaro/src/codyco-superbuild/external/YARP/src/libYARP_logger/include -I/usr/include/python2.7 -o /Users/traversaro/src/codyco-superbuild/build/external/YARP/lib/python/yarpPYTHON_wrap.cxx /Users/traversaro/src/codyco-superbuild/external/YARP/bindings/yarp.i
~~~

charleswilmot, 2016-12-14:
Hello @traversaro ,

First of all thank you for your precious help!

Here it is:

```
[100%] Swig source
cd /home/wilmot/.software/yarp/build/bindings && /home/wilmot/.software/CMake/bin/cmake -E make_directory /home/wilmot/.software/yarp/build/lib/python
cd /home/wilmot/.software/yarp/build/bindings && /usr/bin/swig2.0 -python -Wall -module yarp -threads -outdir /home/wilmot/.software/yarp/build/lib/python -c++ -I/usr/include/opencv -I/usr/include -I/home/wilmot/.software/yarp/build/generated_include -I/home/wilmot/.software/yarp/src/libYARP_OS/include -I/home/wilmot/.software/yarp/src/libYARP_sig/include -I/home/wilmot/.software/yarp/src/libYARP_eigen/include -I/home/wilmot/.software/yarp/src/libYARP_gsl/include -I/home/wilmot/.software/yarp/src/libYARP_math/include -I/home/wilmot/.software/yarp/src/libYARP_dev/include -I/home/wilmot/.software/yarp/src/libYARP_name/include -I/home/wilmot/.software/yarp/src/libYARP_manager/include -I/home/wilmot/.software/yarp/src/libYARP_logger/include -I/usr/include/python3.4m -o /home/wilmot/.software/yarp/build/lib/python/yarpPYTHON_wrap.cxx /home/wilmot/.software/yarp/bindings/yarp.i
```

traversaro, 2016-12-14:
The problem is that the `OpenCVConfig.cmake` of OpenCV 2.4 calls directly `include_directories(${OpenCV_INCLUDE_DIRS})` instead of simply providing the `OpenCV_INCLUDE_DIRS` variable to the users. This has been solved in OpenCV 3, but for the time being we need a workaround in YARP. 

https://github.com/opencv/opencv/blob/2.4/cmake/templates/OpenCVConfig.cmake.in#L123

traversaro, 2016-12-14:
@charleswilmot this PR https://github.com/robotology/yarp/pull/1025 should solve your issue, can you test it?

charleswilmot, 2016-12-14:
@traversaro I tested it, but still have the exact same error.

traversaro, 2016-12-14:
@charleswilmot Can you copy paste the same lines of the output you copy-pasted before? If the `-I/usr/include/opencv -I/usr/include` part disappeared, that means that SWIG itself search for system paths before user provided ones, and in that case I am afraid that uninstall the system YARP is the only solution.

charleswilmot, 2016-12-15:
Once ore, thank you very much for your help @traversaro.
Here are the two lines:
```
[100%] Swig source
cd /home/wilmot/.software/yarp/build/bindings && /home/wilmot/.software/CMake/bin/cmake -E make_directory /home/wilmot/.software/yarp/build/lib/python
cd /home/wilmot/.software/yarp/build/bindings && /usr/bin/swig2.0 -python -Wall -module yarp -threads -outdir /home/wilmot/.software/yarp/build/lib/python -c++ -I/usr/include/opencv -I/usr/include -I/home/wilmot/.software/yarp/build/generated_include -I/home/wilmot/.software/yarp/src/libYARP_OS/include -I/home/wilmot/.software/yarp/src/libYARP_sig/include -I/home/wilmot/.software/yarp/src/libYARP_eigen/include -I/home/wilmot/.software/yarp/src/libYARP_gsl/include -I/home/wilmot/.software/yarp/src/libYARP_math/include -I/home/wilmot/.software/yarp/src/libYARP_dev/include -I/home/wilmot/.software/yarp/src/libYARP_name/include -I/home/wilmot/.software/yarp/src/libYARP_manager/include -I/home/wilmot/.software/yarp/src/libYARP_logger/include -I/usr/include/python3.4m -o /home/wilmot/.software/yarp/build/lib/python/yarpPYTHON_wrap.cxx /home/wilmot/.software/yarp/bindings/yarp.i
```
The `-I/usr/include/opencv -I/usr/include` is still here

traversaro, 2016-12-15:
Good news, then we still have hope to easily fix the issue. : ) 
Can you try with a clean cmake build of that PR? Note that the PR  does not have been merged, so you should switch to the `fix-opencv24-pollution` branch to test it.

charleswilmot, 2016-12-15:
So I deleted the repository from my hard drive and cloned only the branch
```
git clone -b fix-opencv24-pollution --single-branch https://github.com/robotology/yarp.git
```
Then I configured the compilation with ccmake (same procedure as in the first post), and finally got it compiled!
```
[100%] Built target _yarp_python
```
I found the files `_yarp.so` and `yarp.py` in the directory `/yarp/build/lib/python/`, so I copied and pasted the file `example.py` in that directory and ran `python3 example.py` which lead to a `Segmentation fault`.
Importing `yarp` or `_yarp` from the python interpreter gave the same result.
I'm not sure where the seg fault comes from...

```
wilmot@lime:~/.software/yarp/build/lib/python$ python3 example.py
Segmentation fault

wilmot@lime:~/.software/yarp/build/lib/python$ python3
>>> import yarp
Segmentation fault

wilmot@lime:~/.software/yarp/build/lib/python$ python3
>>> import _yarp
Segmentation fault
```

charleswilmot, 2016-12-19:
I found out why I got this segmentation fault. I had to use an other version of SWIG

Thank you very much for you help!

closed 2016-12-19 by charleswilmot