robotology/yarp #1028: Value val == 0 gives SEGFAULT

gmmarconi, 2016-12-15:
```
    Value val = 1000;
    if (val == 0)
```
The == operator for values implicitly assumes that the right operand is a string, thus this kind of code gives segmentation fault.

C.f.r. line 268 of `yarp::os::Value::operator==`
```
 // comparisons with strings worked "accidentally", users depend on them
    bool operator==(const char *alt) const {
        return asString() == alt;
    }
```
This behaviour emerged using `bottle.get(n)` without `.asInt()`

randaz81, 2016-12-15:
1) Can't this be fixed by defining additional overloads? Compiler should be smart enough to choose the right one. For example:
```
bool operator==(int &alt) const;
bool operator==(double &alt) const;
```

2) Btw, since we have devel and master branches, it would be nice to deprecate this $%#@ in the next yarp release. In this way, users could have six months of time to fix their "accidentally working" apps.

aerydna, 2016-12-21:
fixed by #1043

PR #1043 (2017-01-10): Deprecate yarp::os::Value::operator==(const char *alt)
fixes #1028.
`<yarp::os::Value> == <literal integer>` calls the `const char*` comparison overload (often resulting in a segfault). new `int` comparison overload has been added with relative test.
closed 2017-03-28 by drdanz