robotology/yarp #1029: cannot find icub_right_arm.xml and icub_left_arm.xml on iCubDarmastadt01

giuliavezzani, 2016-12-16:
Hi all, 

I'm looking for the following files:

https://github.com/robotology/robots-configuration/tree/master/iCubGenova01/hardware/motorControl

but for the `iCubDarmstadt01`, in order to perform [**arm fine calibration**](http://wiki.icub.org/wiki/ArmFineCalibration) and update the **zeros** values.

I couldn't find any files like those and, I didn't find even any file where the **zeros** for the robot are set.

@paraschos suggested to ask @traversaro for any suggestions.

Thanks,
Giulia

traversaro, 2016-12-16:
I think we can move this issue to `icub-support`, right?

traversaro, 2016-12-16:
Issue moved to [robotology/icub-support #340](https://github.com/robotology/icub-support/issues/340) via [**ZenHub**](https://www.zenhub.com/)

closed 2016-12-16 by traversaro