robotology/yarp #1032: Error in `FrameGrabberControl2Impl.h`

gregoire-pointeau, 2016-12-19:
Hello, I try to compile a module in the repository `wysiwyd` : the module `autobiographicalMemory` and I get a compilation error that seems to come from yarp:

>Error	1	error C2332: 'struct' : missing tag name	C:\Robot\yarp\src\libYARP_dev\include\yarp\dev\FrameGrabberControl2Impl.h	63	autobiographicalMemory (AutoBiographicalMemory\autobiographicalMemory)
Error	2	error C2144: syntax error : 'yarp::dev::<unnamed-tag>' should be preceded by ')'	C:\Robot\yarp\src\libYARP_dev\include\yarp\dev\FrameGrabberControl2Impl.h	63	autobiographicalMemory (AutoBiographicalMemory\autobiographicalMemory)
Error	3	error C2144: syntax error : 'yarp::dev::<unnamed-tag>' should be preceded by ';'	C:\Robot\yarp\src\libYARP_dev\include\yarp\dev\FrameGrabberControl2Impl.h	63	autobiographicalMemory (AutoBiographicalMemory\autobiographicalMemory)
Error	4	error C2059: syntax error : ')'	C:\Robot\yarp\src\libYARP_dev\include\yarp\dev\FrameGrabberControl2Impl.h	63	autobiographicalMemory (AutoBiographicalMemory\autobiographicalMemory)
Error	5	error C2027: use of undefined type 'yarp::dev::<unnamed-tag>'	C:\Robot\yarp\src\libYARP_dev\include\yarp\dev\FrameGrabberControl2Impl.h	63	autobiographicalMemory (AutoBiographicalMemory\autobiographicalMemory)

`yarp` `icub-main` and `wysiwyd` are up to date on master branches. Did someone saw already this problem?

barbalberto, 2016-12-19:
Hi, can you please say something more about your comiplation set-up?
- Operative system
- Compiler (which VS if win)
- branch of yarp / icub
- last update of repo?

This is too few information to actually start debugging

gregoire-pointeau, 2016-12-19:
OS: WIndows 10;
Compiler: VS 12 - 2013  CMake 3.5.2
branches: `master`
update this morning (Monday 19/12/2016)

barbalberto, 2016-12-22:
Can any windows user check this?
@pattacini @randaz81

drdanz, 2017-01-09:
Ping @pattacini @randaz81, can you please test?

pattacini, 2017-01-09:
It was on my backlog but I'm currently very busy higher priority issues, sorry guys.

I think we could assign it to one of @robotology/ifi-padawans