robotology/yarp #1063: yarprun assertion failure on closure

randaz81, 2017-01-13:
yarprun prints:

`[FATAL]Assertion failure at C:` (incomplete string!)

when I try to terminate it with `CTRL+C`. 
Observed on devel branch, WIN32, YARP_EXPERIMENTAL_CXX11=false

randaz81, 2017-01-15:
Expection is triggered inside `Network.cpp`
```
static void removeNameSpace()
{
    if (__multi_name_space != YARP_NULLPTR) {
        delete __multi_name_space;
        __multi_name_space = YARP_NULLPTR;
    }
}
```
being __multi_name_space an invalid pointer.

drdanz, 2017-01-17:
I think that @francesco-romano had a similar issue with the `__multi_name_space` variable, but I don't remember the details, perhaps they are related?

francesco-romano, 2017-01-17:
If I am not mistaken it was not related to the `__multi_name_space` variable (that was another issue, unsolved) but to `yarprun` in general. 
This is the related [issue](https://github.com/robotology/yarp/pull/865).

drdanz, 2017-01-18:
Hmm, I think we are talking about 2 different issues, I was referring to the one related to matlab not unloading the dll, and the yarp_auto_init stuff...

francesco-romano, 2017-01-18:
Ok, but that was related to static variables initialization. I didn't think it related to this.
But we can meet face 2 face (with @randaz81  too) and debug the problem together

drdanz, 2017-10-17:
Does this happen with VS2015 as well?
Perhaps related with #1061 and #1035

Nicogene, 2018-06-07:
It has been fixed by #1623 ?