robotology/yarp #1068: Memory Leak in PolyDriverTest

Nicogene, 2017-01-17:
Running Valgrind on `harness_dev regression PolyDriverTest` I encountered this memory leak, I suppose that it is related to the test on `controlboardwrapper2`:

``` 
==5675== Thread 13:
==5675== Syscall param write(buf) points to uninitialised byte(s)
==5675==    at 0x68A16FD: ??? (syscall-template.S:84)
==5675==    by 0x6C19FB6: ACE::send_n_i(int, void const*, unsigned long, unsigned long*) (in /usr/lib/libACE-6.3.3.so)
==5675==    by 0x4F559EB: ACE::send_n(int, void const*, unsigned long, ACE_Time_Value const*, unsigned long*) (ACE.inl:188)
==5675==    by 0x4F55B46: ACE_SOCK_Stream::send_n(void const*, unsigned long, ACE_Time_Value const*, unsigned long*) const (SOCK_Stream.inl:119)
==5675==    by 0x4F562D6: yarp::os::impl::SocketTwoWayStream::write(yarp::os::Bytes const&) (SocketTwoWayStream.h:144)
==5675==    by 0x4F29125: yarp::os::impl::BufferedConnectionWriter::write(yarp::os::OutputStream&) (BufferedConnectionWriter.cpp:165)
==5675==    by 0x4F19757: yarp::os::AbstractCarrier::write(yarp::os::ConnectionState&, yarp::os::SizedWriter&) (AbstractCarrier.cpp:176)
==5675==    by 0x4FC4C48: yarp::os::impl::Protocol::write(yarp::os::SizedWriter&) (Protocol.cpp:419)
==5675==    by 0x4FB2075: yarp::os::impl::PortCoreOutputUnit::sendHelper() (PortCoreOutputUnit.cpp:292)
==5675==    by 0x4FB0773: yarp::os::impl::PortCoreOutputUnit::run() (PortCoreOutputUnit.cpp:71)
==5675==    by 0x5001D6A: theExecutiveBranch(void*) (ThreadImpl.cpp:104)
==5675==    by 0x6C71F55: ACE_OS_Thread_Adapter::invoke() (in /usr/lib/libACE-6.3.3.so)
==5675==  Address 0xadd6af0 is 0 bytes inside a block of size 128 alloc'd
==5675==    at 0x4C2E80F: operator new[](unsigned long) (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==5675==    by 0x4F5A3B0: yarp::os::ManagedBytes::allocate(unsigned long) (ManagedBytes.cpp:70)
==5675==    by 0x420B07: yarp::sig::VectorOf<double>::VectorOf() (Vector.h:120)
==5675==    by 0x5A740DC: jointData::jointData() (jointData.h:39)
==5675==    by 0x5A78ECA: yarp::os::PortWriterBufferAdaptor<jointData>::PortWriterBufferAdaptor(yarp::os::PortWriterBufferManager&, void*) (in /home/ngenesio/yarp/build/lib/libYARP_dev.so.2.3.69.4)
==5675==    by 0x5A785D8: yarp::os::PortWriterBuffer<jointData>::create(yarp::os::PortWriterBufferManager&, void*) (PortWriterBuffer.h:114)
==5675==    by 0x4FB9AC7: PortWriterBufferBaseHelper::get() (PortWriterBuffer.cpp:79)
==5675==    by 0x4FB9571: yarp::os::PortWriterBufferBase::getContent() (PortWriterBuffer.cpp:190)
==5675==    by 0x5A75889: yarp::os::PortWriterBuffer<jointData>::get() (PortWriterBuffer.h:147)
==5675==    by 0x5A5C94F: yarp::dev::ControlBoardWrapper::run() (ControlBoardWrapper.cpp:876)
==5675==    by 0x4FC7736: RateThreadCallbackAdapter::singleStep() (RateThread.cpp:160)
==5675==    by 0x4FC7931: RateThreadCallbackAdapter::run() (RateThread.cpp:186)
==5675== 
```
I will try to fix it, any idea or suggestion is more than welcome! :wink:

drdanz, 2017-01-17:
@Nicogene I think that the leak was introduced in this commit range: `d2e1e6b58d9aaa14257ad20d94260ac473b72d03...79ae629cc8cf5257f07744c5f28802530382f9c2`

The travis build did not run the tests for a few days for a bug in the `.travis.yml file, and someone sneakily introduced this leak.
I suggest you run `git bisect` to find the broken commit, this will make it easier to fix it.

Let me know if you need help!

drdanz, 2017-01-17:
This should help:
```bash
git log --pretty=oneline --no-merges d2e1e6b58d9aaa14257ad20d94260ac473b72d03...79ae629cc8cf5257f07744c5f28802530382f9c2 ^origin/master
```

Nicogene, 2017-01-19:
Solved by #1069.

closed 2017-01-19 by Nicogene