robotology/yarp #1074: Yarpview and Yarpmotorgui do not start on fresh windows installation (Qt problem)

lorejam, 2017-01-25:
Yarp 2.3.66 installed from pre-compiled binaries (MSVC12 amd64 installer, found here: http://www.yarp.it/installation_downloads.html) on Windows 10 Home 64-bits.
Yarp check works fine.

Yarpview does not start, and gives the following errors (same for Yarpscope):

yarpview.exe - Entry Point Not Found
---------------------------
The procedure entry point ?disableReset@QNonContiguousByteDevice@@QEAAXXZ could not be located in the dynamic link library C:\Users\Lorenzo\Documents\SW\robotology\Qt-5.4.1_v12_x86_amd64\bin\Qt5Network.dll. 

and

yarpview.exe - Entry Point Not Found
---------------------------
The procedure entry point ?tryShortcutEvent@QShortcutMap@@QEAA_NPEAVQObject@@PEAVQKeyEvent@@@Z could not be located in the dynamic link library C:\Users\Lorenzo\Documents\SW\robotology\Qt-5.4.1_v12_x86_amd64\bin\Qt5Quick.dll. 

Yarpmotorgui does not start, and gives the following error:

yarpmotorgui
---------------------------
This application failed to start because it could not find or load the Qt platform plugin "windows"
in "".
Reinstalling the application may fix this problem.

Any idea?
Many thanks.

Tobias-Fischer, 2017-01-26:
Hi @lorejam, I recently used the Yarp 2.3.68 binaries (https://github.com/robotology/yarp/releases) on a Windows machine and didn't have any issues. You might want to try that version :).

Good luck. Best, Tobi

drdanz, 2017-02-02:
I'm not sure what is happening here, but the http://www.yarp.it/installation_downloads.html page was still pointing to the old release. I just updated it, it will be online in a couple of hours. Please try latest release and let us know if you still have this problem.

drdanz, 2017-02-02:
Also the problem *could* be related to other Qt installations in the PATH, I saw something similar happening with CMake in the PATH, since it installs its own version of Qt5, but not all the plugins and other required stuff
Please check if CMake is in the PATH, and in case ensure that YARP binary dir is **BEFORE** CMake.

kosenhitatchi, 2017-04-21:
Good afternoon everyone.

I'm running yarp on a  Windows 10 Home 64-bits. 
I've installed Yarp 2.3.68 binaries ( for MVS 2013), and Yarp check works fine.

But I'm facing the same problems as @lorejam, I have read the previous reply and checked if CMake was in my PATH ( he's not).

>Qt5_DIR = C:\Program Files\robotology\Qt-5.6.1_v12_x86_amd64\lib\cmake\Qt5
INCLUDE = ... ; C:\Program Files\robotology\Qt-5.6.1_v12_x86_amd64\include
PATH : C:\Program Files\robotology\Qt-5.6.1_v12_x86_amd64\bin; ... (No CMake elements here)

But even if my variables are pointing to the right folder (C:\Program Files\robotology\Qt-5.6.1_v12_x86_amd64), I'm still facing this issue.

As an example : If I run 

> yarpview

I'm getting : _Entry Point Not Found_
> The procedure entry point ?write@QWindowsPipeWriter@@QEAA_NAEBVQByteArray@@@Z could not be located in the dynamic link library C:\Program Files\robotology\Qt-5.6.1_v12_x86_amd64\bin\Qt5Network.dll.

I've checked if "Qt5Networkd.dll" is at the location mentioned above, He is.

Do you have any solution or idea to solve this Qt problem ?

Sincerely

lornat75, 2018-02-08:
I am having the same problem. Trying again using Qt from Qt's installer (instead of the one shipped with the YARP binaries); Visual Studio 12 2013, YARP 2.3.70.

claudiofantacci, 2018-02-08:
I have Windows 10, YARP `2.3.71` (devel branch), Qt `5.8-6` installed with [`vcpkg`](https://github.com/Microsoft/vcpkg) and Visual Studio 2017 (release 15.4.5, toolset v141) and I don't have this problem.

traversaro, 2018-02-08:
@lornat75 
What is your `PATH` env variable ?
What is the path of the dll printed in the `The procedure entry point` error? 
What is your `Qt5_DIR` CMake variable in the CMake cache of YARP?

traversaro, 2018-02-08:
For debugging this kind of problems the Dependency Walker is typically useful: http://www.dependencywalker.com/ .

claudiofantacci, 2018-02-08:
> For debugging this kind of problems the Dependency Walker is typically useful: http://www.dependencywalker.com/ .

This project took back dependecy walker (discontinued 2006) and made it fresh new: https://github.com/lucasg/Dependencies

I use it regularly and may be of help.

lornat75, 2018-02-08:
In the PATH my robotology/qt installation (i.e. the version installed by the YARP binaries) is first, and the error points to those DLLs. Therefore it does not seem a problem of DLL clash.

traversaro, 2018-02-08:
> Therefore it does not seem a problem of DLL clash. 

It still possible that CMake found another Qt installation, and is using those headers for the compilation (the `Qt5_DIR` cmake variable should contain this information).

lornat75, 2018-02-08:
I checked CMake variables, Qt5_DIR points to my robotology Qt installation.

lornat75, 2018-02-08:
Solved. It was a PATH problem (MikTeX adding its own installation of some qt dlls in the PATH). I did not notice it, because I was checking only the user PATH and not the system PATH variables in windows. 

The error was misleading because MikTeX was hijacking only some of the DLLs causing the missing symbol error in another DLL.

Yet another example of DLL hell. 

@claudiofantacci @traversaro thanks for the help, and in particular for pointing out the new version of dependency walker.

closed 2018-02-08 by lornat75