robotology/yarp #1081: Same folder used for plugins and CMake config files

drdanz, 2017-02-02:
Given that

1. CMake config files are installed in `<prefix>/lib/YARP`
2. Plugin libraries are installed in `<prefix>/lib/yarp`
3. On macOS, the file system is case insensitive.

This means that on macOS both CMake files and plugin libraries are installed in the same folder.

This is not optimal. Also on case sensitive file system it is not really nice to have `/usr/lib/YARP` and `/usr/lib/yarp`

I suggest to change CMake folder to one of:

* `<prefix>/lib/cmake/YARP`
* `<prefix>/share/yarp/cmake` (this folder is already used for CMake modules installed by yarp)
* `<prefix>/share/yarp` (this is the folder used by vcpkg on windows)

Also, since CMake v3.7.0, on windows, the same paths are supported, therefore we might consider changing the folder on windows as well (at the moment `<prefix>/cmake`), maybe adding a check on CMake version.

Anyone against this change?

cc @francesco-romano, @lornat75, @traversaro

cc also @mbrunettini since this change will have an impact on packaging

francesco-romano, 2017-02-02:
I agree with the change.

Just as a side note:
the file system (HFS+) on macOS may be case sensitive. It depends on how the user format the drive.
Additionally, macOS is starting to migrate to the new [Apple File System (AFS)](https://developer.apple.com/library/content/documentation/FileManagement/Conceptual/APFS_Guide/Introduction/Introduction.html)

traversaro, 2017-02-02:
+1 

Just a note: I think that in vcpkg you are free to put whatever you want in `<prefix>/share/<portname>`, so also `<prefix>/share/yarp/cmake` will be ok. 

Interestingly the `<prefix>/lib/cmake/<packagename>` is the one used in the CMake documentation (
https://cmake.org/cmake/help/v3.7/manual/cmake-packages.7.html) even if I don't think they endorse it in any way as the "official" one.

drdanz, 2017-02-02:
My personal preference is that `${CMAKE_INSTALL_LIBDIR}/cmake/<name>^` (i.e. `/usr/lib/<triplet>/cmake/YARP`) should be used for architecture dependent packages (i.e. "normal libraries") so that it works with "multiarch", and `${CMAKE_INSTALL_DATADIR}/<name>^/cmake` (i.e. `/usr/share/YCM/cmake`) should be used for architecture independent packages (i.e. header only libraries and cmake only packages).

`^` Note that `<name>` is case insensitive

traversaro, 2017-02-02:
@drdanz I totally agree, see https://github.com/Microsoft/vcpkg/issues/77#issuecomment-255862551 .

PR #1451 (2017-11-27): Change YARP CMake file destination on Linux/macOs to "lib/cmake/YARP"
Fixes #1081 

As discussed with @mbrunettini, we will merge this when the release scripts are updated as well.

@francesco-romano can you check if brew formulas need to be updated?
commit 41ebaaa5 (2017-11-27): Change YARP CMake file destination on Linux/macOs to "lib/cmake/YARP"

Fixes #1081
closed 2018-02-12 by mbrunettini