robotology/yarp #1084: [iCubDarmstadt01] Yarpmanager cannot launch GUIs

giuliavezzani, 2017-02-03:
If I try to launch **GUIs**  (`yarpview`, `skinManagerGui` etc) from the **yarpmanager** on `icub19` of the `iCubDarmstadt01`,  they cannot be launched. If I launch from terminal instead, everything works fine.

@Nicogene suggested that there are some problems with `qt`, but we cannot check properly or install any new version of `qt` because internet is not working on the robot laptops.

closed 2017-02-03 by giuliavezzani

drdanz, 2017-02-03:
@giuliavezzani How did you solve this? Please leave some comment when you close an issue to explain why you are doing that.

giuliavezzani, 2017-02-06:
You are right. Actually I didn't do anything on purpose for solving the issue. The yarpmanager  simply started to work properly. 
The only difference might be that I installed some libraries, that I need for my code .. But I am know sure the two facts are correlated.