robotology/yarp #1085: ACE_HAS_STRING_HASH check in FindACE.cmake leads to compilation failure when switching between different ACE versions

PeterBowman, 2017-02-06:
Tested on Windows 10 64 bits, Visual Studio 2015, CMake 3.7.0 (GUI). Steps to reproduce this issue:

1. Download and install the latest YARP binaries (currently 2.3.68) in the usual location (*C:\Program files*) so that *FindACE.cmake* will pick it in a later step. ACE (6.3.4) should be installed along with YARP.
2. Clone/download YARP repository from GitHub (`master` branch).
3. Download ACE from [here](http://download.dre.vanderbilt.edu/) (*ACE.zip*, current version: 6.4.2).
4. Build ACE ([instructions](http://www.dre.vanderbilt.edu/~schmidt/DOC_ROOT/ACE/ACE-INSTALL.html#msvc)).
5. Open CMake GUI, provide path to cloned/non-installed YARP as the directory where source code is located, provide path to build directory where binaries will be built, click *Configure*.
6. Find CMake entries starting with `ACE_`, all of them should point to the installation path of YARP (*C:\Program files\robotology*) as explained in the 1st step.
7. Find the `ACE_HAS_STRING_HASH:INTERNAL` entry in *CMakeCache.txt*, it should be set to the empty string according to the `check_cxx_source_compiles()` test performed in *FindACE.cmake* for this version of ACE.
8. Point `ACE_` entries to the paths that correspond to your built version of ACE, click *Configure* again, then *Generate* and open the Visual Studio solution.
9. Select the *Release* configuration and build just YARP_OS. You are expected to see several C2766 errors originating in [*PlatformMap.h*](https://github.com/robotology/yarp/blob/e122128/src/libYARP_OS/include/yarp/os/impl/PlatformMap.h) with a description similar to: `explicit specialization; 'ACE_Equal_to<std::string>' has already been defined`.

Causes:

* CMake's [`check_cxx_source_compiles()`](https://cmake.org/cmake/help/latest/module/CheckCXXSourceCompiles.html) command stores the result variable (`ACE_HAS_STRING_HASH`) in the cache. More importantly, consequent invocations of this command do not perform the check again because of a restrictive `if(NOT DEFINED ACE_HAS_STRING_HASH)` clause (see [source code](https://github.com/Kitware/CMake/blob/a00cca9/Modules/CheckCSourceCompiles.cmake#L31)). In other words, *FindACE* fails to check the availability of certain features when distinct versions of ACE are supplied to CMake during the configuration phase.

Workarounds:

1. Open *PlatformMap.h* and invert the `#ifndef ACE_HAS_STRING_HASH` clause by creating/deleting the symbol, replacing with `#ifdef`, etc.
2. Move the installed YARP folder out of the standard paths so that `find_package()` is unable to locate it (through *FindACE.cmake*) upon clicking *Configure* for the first time.
3. Remove the `ACE_HAS_STRING_HASH` entry from *CMakeCache.txt*, restart CMake GUI (it stores an internal non-persistent cache that must be emptied) and configure again.

Proposed solution:

* Delete the `ACE_HAS_STRING_HASH` variable from cache (e.g. `unset(ACE_HAS_STRING_HASH CACHE)`) prior to calling `check_cxx_source_compiles()` in *FindACE.cmake*. Additional temporary, internal variables may be used to prevent from performing this check for the same ACE version each time *Configure* is pressed (that is, unset the variable only when switching between different ACE versions).

Motivation:

* One could want to keep a separate copy of YARP + dependencies for testing purposes or compilation of additional features (e.g. YARP bindings or plugins). In such cases, users of downstream software are presented a quite hackish solution to solve the problem: [example](https://github.com/asrob-uc3m/robotDevastation-user-manual/commit/45476da#diff-2d8b21e461df0f6cd718dc1699a64ac5). It might be better to include a simple workaround as the proposed one to handle this automatically.

Somehow related issues: #202, #248.

PeterBowman, 2017-07-24:
47a8de0 could have fixed this, I'll check it later.

drdanz, 2017-08-07:
@PeterBowman I think this is fixed, please reopen if you still have this issue

closed 2017-08-07 by drdanz

PeterBowman, 2017-08-19:
Confirmed, sorry for the delay.