robotology/yarp #1087: Home in yarpmotorgui does not use anymore the configuration file

francesco-romano, 2017-02-08:
Since the new version on devel, the yarpmotorgui does not use anymore the zeros defined in the configuration file passed to the app through `--from` parameter.

It looks like that the part name saved in each `PartItem` class contains a starting slash , e.g. `/right_arm`.

As a workaround I added to [this part](https://github.com/robotology/yarp/blob/master/src/yarpmotorgui/partitem.cpp#L1096) and to [this](https://github.com/robotology/yarp/blob/master/src/yarpmotorgui/partitem.cpp#L831) the following check:
```cpp
QString zero = QString("/%1_zero").arg(m_partName);
    if (zero.startsWith("/")) {
        zero.remove(0, 1);
    }
```
but of course this is just a workaround and the code should be fixed correctly when the part is saved.

cc @randaz81 @aerydna

randaz81, 2017-02-08:
yes, actually the configuration file is going to disappear completely, so there will be a different way to set the zeros..

barbalberto, 2017-02-08:
For calibration and zeros the motorgui uses the values set in the calibrator. The gyu simply ask to the robotInterface to call the calibration / homing routine of the calibrator.
If the guy is complaining that it is not able to do it, then there is a wrong configuration of the robotInterface.
This workaround is discouraged, please fix the robotInterface config file instead.

Therefore I'd not mark this as a bug, it is an intended behaviour

francesco-romano, 2017-02-08:
@barbalberto actually we need to have the ability to have multiple zeros for the robot after it has been started up (so I assume this should not go in the calibration).

If the configuration files are going to disappear (as long as we have an alternative) it is fine. But in the meantime I still think this is a bug or something not documented, and considering that we need to use the robot, imho this should be fixed (or the alternative be provided and documented)

traversaro, 2017-02-08:
I want to add that our use of the `yarpmotorgui` to set the robot to known whole-body position is probably suboptimal, and we will eventually  need to a move to some ad-hoc tool for that. 

> This workaround is discouraged, please fix the robotInterface config file instead.

We have those configuration files on a repository (`codyco-modules`) that until now was able to work successfully with both yarp `master` and `devel` successfully. Fixing the file(s) would mean that we need to branch also that repo in `master` and `devel`, that we would like to avoid (if possible).  Avoiding such strong dependency of the downstream code on the particular yarp version is exactly the reason we have a deprecation policy (https://github.com/robotology/yarp/blob/master/.github/CONTRIBUTING.md#deprecated-features). 
The policy is not extremely detailed on configuration files, so I think it is reasonable that for those we reason in a case-by-case way. However changing the configuration file format of one of the most used tools in yarp without any notice, documentation or ChangeLog update is at least a *documentation bug*, so I think it safe to keep this issue as a *bug*. : )

randaz81, 2017-02-09:
I was unware of the files contained in https://github.com/robotology/codyco-modules/tree/master/src/modules\torqueBalancing\app\robots.
I could merge you patch, howeverI definitely think that yarpmotorgui is not the right tool to do what you want to do. And since I planned to the remove the .ini files when the devel will be merged in master, I think it's a more efficient use of time of I help you in finding a different solution. Can we discuss this tomorrow? 

By the way zeros are not the only problem. Let's consider the current status of your files:
```//name of the robot          
name icubSim
//parts to be opened by the GUI            
parts (head torso left_arm right_arm right_leg left_leg)
```
`name` is wrong. It should be `robot` instead.
`name` is used to specify the full port of a robot. You can use one of the two following syntaxes:

1)

```      
robot icubSim   
parts (head torso left_arm right_arm right_leg left_leg)
```
OR: 
2)

```       
name(/icubSim/head /icubSim/torso /icubSim/left_arm /icubSim/right_arm /icubSim/right_leg /icubSim/left_leg)
```

So I'm not sure about what the parser is currently doing... (probably untested, I'll check).
PS1: option 2 is unmerged... because of the winterschool!
PS2: there exist an additional command --skip_parts to remove parts detected by the descriptionServer

francesco-romano, 2017-02-09:
@randaz81 definitely we would like to have a chat f2f

Furthermore, it was not my intention to propose this as a patch, as for sure it is just a workaround (but probably a `PartItem` should not start with a "/").
I know that sometimes changes are necessary. Maybe something more gradual, less disruptive and with a documented migration to the new tools?

barbalberto, 2017-02-09:
Let's have a meeting because answering here is too long.

Short version is: user _must not_ have access to home button configuration, for doing what you need the the solution is the sequence tab.

Removing this possibility is a bugFIX of a long standing error in the motorgui. So basically this request is:
"A bug has been fixed, but I was so used to it, please re-introduce it".

We can talk about adding new functionalities in case sequence file not is not enough.

traversaro, 2017-02-09:
> Short version is: user must not have access to home button configuration, for doing what you need the the solution is the sequence tab.

Unless this was written somewhere, I fail to see how the user should have been aware of this.

francesco-romano, 2017-02-13:
In the meantime that a proper solution is found and implemented, I think some sort of workaround should be committed to the repository as this issues is causing problems to other people too.

@randaz81 @barbalberto @traversaro @lornat75

lornat75, 2017-02-14:
I agree. @randaz81 @barbalberto can you roll-back the changes while you fix the problems encountered by @traversaro and @francesco-romano?

randaz81, 2017-02-17:
Fixed by af7b3e6ba5c0c69b907c8f58be621889f36e9a5a and f5d3b6e5dc9f581d1dcee7b2fb7972092d504560. 
New buttons added:
- "Execute script1"
- "Execute script2"
- "Move to custom position1"
- "Move to custom poisiton2"

Only minor changes to `yarpmotorgui.ini` required, as agreed with @traversaro and @francesco-romano.

yarpmotorgui documentation http://www.yarp.it/yarpmotorgui.html is heavily outdated and requires major changes (TBD)

closed 2017-02-17 by randaz81