robotology/yarp #1089: Reduce wait state in yarpmanager when stopping a subset of modules

pattacini, 2017-02-12:
Hi @apaikan 

@Tobias-Fischer and I have a follow-up to #1051 for the case when **only a subset of modules** is selected to be stopped.

Instead of relying directly on the values specified for the `stop` option within the XML file, it would be better to substract the minimum value among those selected. This way, we are not going to wait unnecessary seconds upon clicking on stop.

#### Example

Say we have 3 selected modules within an application containing many more, and the following settings for the three modules are given in order:

```xml
<ensure>
   <wait when='stop'> 3 </wait> 
</ensure>

<ensure>
   <wait when='stop'> 5 </wait> 
</ensure>

<ensure>
   <wait when='stop'> 10 </wait> 
</ensure>
```

When we stop just the three modules, we could better wait:

- `0` seconds for the first instead of `3`
- `2` seconds for the second instead of `5`
- `7` seconds for the third instead of `10`

We can achieve the above behavior by just subtracting `3` (the minimum) from all of them.
The same should apply also in the case we select only one module: just wait `0` second.

drdanz, 2017-02-16:
@pattacini Is there any use case where you could actually want the module to stop even if it is the only one selected?
Or maybe you want to stop n seconds after a specific module is stopped? Perhaps a syntax like this would be easier to write and to understand?

```xml
<!-- module2 -->
<ensure>
   <wait when='stop' after='module1'> 3 </wait> 
</ensure>

<!-- module3 -->
<ensure>
   <wait when='stop' after='module2'> 2 </wait> 
</ensure>

<!-- module4 -->
<ensure>
   <wait when='stop' after='module3'> 5 </wait> 
</ensure>
```

pattacini, 2017-02-16:
Yes, there're lots of use case when we wanna just stop one module of an application rather than the complete application: think of debugging sessions for example.

The rationale for the structure we came up with is described in #1051. Essentially, there might be multiple dependencies while stopping modules and the design drive was to keep things simple and effective.

pattacini, 2017-09-16:
Task reassigned.

pattacini, 2017-10-13:
@Nicogene do you think you can have a look at this?

Nicogene, 2017-10-18:
Fixed by #1414

closed 2017-10-18 by Nicogene

pattacini, 2017-10-18:
Thank you @Nicogene 👍

PR #1414 (2017-10-18): Improvement yarpmanager
After #1407 , it has been possible to introduces some new nice features:
- `yarp clean` button: before actually performing it, it asks which timeout do you want to use. It fixes #1401 
- `yarp name list` button:  "Ports" has been added in the entities tree, it is populated by this button.
:warning:  at the startup `yarpmanager` executes a `yarp name list` command.

[image]

Moreover it fixes #1089 .
Please review code.

demoRedBall TESTED.