robotology/yarp #1090: Occasional zeros when reading iCubGenova02 IMU

S-Dafarra, 2017-02-13:
When reading linear acceleration components from the IMU on the iCubGenova02 head, occasionally zeros occur. It happens once every couple of minutes in an apparently random way. The X-,Y- and Z- components stay identically equal to zero for about 20-30ms before getting back to "normal" values.

@gabrielenava @traversaro  @francesco-romano  @DanielePucci

traversaro, 2017-02-13:
To clarify: this issue was opened in `yarp` because it seems to be related to the `imuBosch_BNO055` driver, contained in https://github.com/robotology/yarp/tree/master/src/devices/imuBosch_BNO055 .

cc @barbalberto

lornat75, 2017-02-14:
Did you notice this from the data read by the driver or the YARP port?

S-Dafarra, 2017-02-14:
Hi @lornat75 , when reading the port ``/icub/inertial``.