robotology/yarp #1092: Include in thrift has unexpected behaviour

francesco-romano, 2017-02-14:


francesco-romano, 2017-02-14:
cc @drdanz @traversaro

francesco-romano, 2017-02-14:
This is not documented (or clear) on the original thrift page.

But *en passant* it is documented in http://www.yarp.it/thrift_tutorial.html. In particular 
> Included files are looked up in the current directory and by searching relative to the path from which the yarp_idl_to_dir macro is executed

If I understood correctly, adding `include "Base.thrift"` matches the current directory, i.e. it is able to find the include file, but of course the generated cpp file does not work.
On the contrary `include "thrift/Base.thrift"` will match the path relative to the `yarp_idl_to_dir` macro which works both when the thrift compiler is running and in the generated cpp.

@drdanz you can decide if this is the intended behaviour and close this issue or not

drdanz, 2017-02-16:
Looks like a bug to me...