robotology/yarp #1094: FrameGrabbers sliding bars update values only on click

tanismar, 2017-02-15:
Hi all, 
It is a small bug with easy workaround, but a bug nevertheless, so I ll post it: On Framegrabbers, when modifying the value of any parameter, the parameters applied only update when the sliding bar is clicked. Therefore, it works fine if they are modified with the mouse, but if the sliders are modified using the keys (sometimes necessary to get a specific value) they don't update the image results. Clicking on the knob after finding the right value solves the issue, but should be updated automatically. 

Thanks!
Tanis

drdanz, 2017-02-16:
Issue moved to [robotology/icub-main #425](https://github.com/robotology/icub-main/issues/425) via [**ZenHub**](https://www.zenhub.com/)

drdanz, 2017-02-16:
Moved to icub-main since FrameGrabberGui is in that repository

closed 2017-02-16 by drdanz