robotology/yarp #1098: All the device doxygen documentation is missing the device name

traversaro, 2017-02-16:
All the yarp devices present in YARP, even if their parameters are nicely documented with markdown tables, are missing a fundamental information for the users: the yarp device type name, i.e. the parameter that must be passed as the `device` parameter when creating the device with the `yarp::dev::PolyDriver` .

traversaro, 2017-02-16:
cc @claudia-lat

drdanz, 2017-02-28:
Which markdown documentation are you referring to? Are they in the doxygen?

The harness_dev executable is somehow able to create documentation when the `CREATE_PLUGIN_TESTS` flag is enabled (see [1](https://github.com/robotology/yarp/blob/master/tests/libYARP_dev/harness.cpp#L234)), and it generates [this](http://yarp.it/group__opencv__grabber__basic.html) [kind](http://yarp.it/group__portaudio__basic.html) [of](http://yarp.it/group__test__grabber__basic.html) [documentation](http://yarp.it/group__dev__examples.html) that includes the name of the device.
Perhaps we could update the devices tested and update the documentation generated?

traversaro, 2018-05-16:
cc @lrapetti because I think it affects you in your initial exploration. .

@drdanz I mean the documentation of each device as incorporates in the class documentation, see 
for example all the devices listed in http://yarp.it/devel/group__dev__impl__wrapper.html .

For new devices, I tried to follow the following style: 
* document the `devicename` as the first thing in the brief description of the device class 
*  Also have a table with the devicename documented in the extended description of the class.  

See as example  http://www.yarp.it/devel/classyarp_1_1dev_1_1MultipleAnalogSensorsRemapper.html 

The automatically documentation that you mention @drdanz is quite nice, but it has some problems: 
* it does not cover all the devices 
* it is extremely well hidden 
I think it is a nice to have addition, that should be linked by the doxygen detailed description of the relative device.

closed 2021-11-12 by drdanz