robotology/yarp #1101: Include yarp_completion into the binary package

diegoferigo, 2017-02-17:
It would be a helpful inclusion for an installation from precompiled binaries.

francesco-romano, 2017-02-27:
This is done at https://github.com/robotology/homebrew-cask/blob/master/yarp.rb#L71 by the homebrew formula, but we should add the possibility also from custom sources installation

diegoferigo, 2017-02-27:
The package from the ubuntu ppa does not have the completion file. @traversaro please remove the macOS label, I still don't have permissions on this project.

drdanz, 2017-02-27:
It should be installed also with the `make install` command.

diegoferigo, 2017-02-27:
I have no experience in `.deb` packaging, but the `yarp` package doesn't seem to have that file, I just checked again with `dpkg -L yarp`. Instead, installing yarp by sources properly installs the bash completion as @drdanz pointed out. FYI I tested the setup inside [this](https://github.com/diegoferigo/dockerfiles/tree/master/YARP) docker container, you can find there the build script.

drdanz, 2017-02-27:
Actually I don't think it is currently installed, I meant that we should fix the installation to install that as well.

traversaro, 2019-03-13:
Renamed the issue to match @drdanz comment in https://github.com/robotology/yarp/issues/1101#issuecomment-282809148 .

PR #2222 (2020-03-04): Move completion files in data, and install the bash-completion file
:warning: Warning: the WindowsPowerShell and the bash completion files are now in a different directories, some scripts might require to be updated.

* The `scripts/WindowsPowerShell` files are now in `data/WindowsPowerShell`
* The `scripts/yarp_completion` file is now in `data/bash-completion/yarp`
* The `bash-completion` file is now installed automatically (#1101)

Fixes #1101
commit b3c0e98f (2020-03-04): Move completion files in data, and install the bash-completion file

Fixes #1101
closed 2020-03-04 by drdanz