robotology/yarp #1102: yarpserver on computer with multiple network interfaces

francesco-romano, 2017-02-17:
Today I experienced the following:
 a computer has multiple network interfaces.
One of these interfaces is attached to a private network while the other is attached a remote-available network.

The server started up on the private network, with the output 
```
IP address: default
Port number: 10000
yarp: Port /vicon active at tcp://192.168.10.2:10000/

register "/vicon" tcp "192.168.10.2" 10000
   + set "/vicon" ips "10.255.35.126" "192.168.10.2" "127.0.0.1"
```

Of course I needed to connect from an external computer. Setting up my yarp configuration to connect to `10.255.35.126` I was able to see the yarp network (i.e. `yarp name list`).
All the ports were opened on the `192.168.10.2` address and thus I was not able to read them.

I open this issue more as a discussion (what do we expect, do we want to change something) than as a real bug report.

cc @drdanz @lornat75  @traversaro @vtikha

barbalberto, 2017-02-20:
This is a long standing issue. If I understood correctly it is a duplicate of or similar to #143, is it right?
Some discussion were made but no decision were took.

francesco-romano, 2017-02-20:
Yep. Closing this as it is a duplicate of https://github.com/robotology/yarp/issues/143

closed 2017-02-20 by francesco-romano