robotology/yarp #1105: Update yarpmotorgui documentation

traversaro, 2017-02-21:
The `yarpmotorgui` documentation available at http://www.yarp.it/yarpmotorgui.html (can be modified at https://github.com/robotology/yarp/blob/75a62fa0756ae6fade7916d9016bcb832ff9aebb/doc/cmd_yarpmotorgui.dox ) should be updated to match the latest version of the yarpmotorgui . 
Bonus points for documenting the changes in the configuration files in the ChangeLog https://github.com/robotology/yarp/blob/devel/doc/release/v2_3_70.md#yarpmotorgui . 

See https://github.com/robotology/yarp/issues/1087#issuecomment-280677006 .

traversaro, 2017-02-21:
cc @fiorisi .

randaz81, 2017-02-22:
http://www.yarp.it/yarpmotorgui.html should reflect the status of branch `master`, and not `devel`, I think... Any idea about how to handle this issue (different documentation between master and devel branch)?

traversaro, 2017-02-22:
> http://www.yarp.it/yarpmotorgui.html should reflect the status of branch master, and not devel, I think... 

Yes, I removed the reference to that page from the issue.

> Any idea about how to handle this issue (different documentation between master and devel branch)?

The proper way is using doxygen to generate documentation for each version. Quoting @drdanz from http://wiki.icub.org/wiki/YARP_Cleanup#Other_Optional_Cleanup : 
> Instead of just the latest documentation at http://eris.liralab.it/yarpdoc/index.html , we should have it for each version (for example http://eris.liralab.it/yarpdoc/2.3.19/index.html) 

As an example, see https://cmake.org/cmake/help/git-master/index.html and the nice version choice menu up on the left.

jgvictores, 2017-02-28:
I'd also point to: http://wiki.icub.org/wiki/Force_Control#Available_control_modes
which has a great colorful description on control modes I'm not sure is available elsewhere.

Updates should include some info on [interaction modes](http://www.yarp.it/classyarp_1_1dev_1_1IInteractionMode.html) which are present in newer yarpmotorgui versions.

drdanz, 2017-03-10:
👍 For @traversaro proposal about the versioned documentation.
Unfortunately I asked for that already thousands of times...

traversaro, 2017-03-30:
Related: https://github.com/robotology/yarp/issues/1154 .

jgvictores, 2017-05-17:
Related: #1139

traversaro, 2017-11-07:
The documentation is now versioned: 
* `master`: http://www.yarp.it/ 
* `devel`: http://www.yarp.it/devel/

jgvictores, 2019-04-22:
> Updates should include some info on [interaction modes](http://www.yarp.it/classyarp_1_1dev_1_1IInteractionMode.html) which are present in newer yarpmotorgui versions.

Some documentation of Interaction Modes at: http://wiki.icub.org/wiki/Control_Modes