robotology/yarp #1113: Compiling YARP bindings for use with C#

KatePaulinne, 2017-02-26:
Hi All,
 
I'm new to iCub and YARP software, and currently trying to compile bindings so that I can use it with C# or MATLAB. I am using YARP 2.3.66 and iCub 1.4.0 installed from pre-compiled binaries. When I try to compile the bindings with CMake I get an error message saying:
 
"Could not find a configuration file for package "YARP" that is compatible
  with requested version "".
The following configuration files were considered but not accepted:
 
    C:/Program Files/robotology/yarp-2.3.66/cmake/YARPConfig.cmake, version: 2.3.66 (64bit)"
 
I was wondering if this may be due to installing different version of YARP previously that has been uninstalled?
 
I was trying to look in the CMakeLists file but can't figure out why exactly it is failing. Sorry if that question was already asked, I did find a few similar questions but none of them really helped me to figure out what's going wrong..
 
I was also trying to point CMake to specific YARP path on C drive, however this failed as well.
 
I would be grateful for any help with this.

Thanks,
Kate

drdanz, 2017-02-27:
Hello Kate,

Are you sure that you are generating the CMake project using the right CMake generator? This kind of error appears when YARP could not be found or when the version(s) found is(are) not compatible with the current project.

Depending on which version you downloaded of the YARP package, you have to use the same CMake generator, check both visual studio version and 32/64 bit.

Please, let us know if this solves the problem or if you still have this issue...

KatePaulinne, 2017-02-28:
Hi, thank you for help, it solved my problem. It now seems all I had to change was to refer CMake to use visual studio 2013 64 bit executable rather than 32, which is why it was failing.

closed 2017-02-28 by KatePaulinne

drdanz, 2017-02-28:
Glad we could help!
I'm not sure how much the C# bindings have been tested, so feel free to report any issue you should find.