robotology/yarp #1114: RateThread does not exit immediately when close is called

drdanz, 2017-02-27:
If a `RateThread` has a very long rate (i.e. `30000 ms`) and a very short execution time, I would expect that, when the thread is stopped, if it is not in the `run()` method, the thread was stopped immediately (or after a short amount of time).
Instead, the thread stops only when it is the next `run()` time, even if `run()` is not actually called.

This probably means that inside `RateThread` implementation there is a blocking `sleep()` of the whole duration of the rate.
I suggest we split this sleep in shorter sleeps and check if it should exit at each interruption.

@lornat75 what do you think?

francesco-romano, 2017-02-27:
Functions like [`sleep`](https://linux.die.net/man/3/sleep) may be interrupted. 
Do we have portable ways to implement this behaviour? I think it is cleaner than multiple sleep calls

drdanz, 2017-02-27:
I think that we want to interrupt only the sleep and not if `run()` function is being executed, so I don't think that sending an interrupt to the thread is a good idea...
Also `sleep()` implementation is very platform dependent and I don't think it's worth investigating how to do this on every platform.
Maybe we could use a condition variable + `wait_for` instead?

francesco-romano, 2017-02-27:
I was thinking to the condition variable too. 
Simply I'm still "missing" how to achieve both without using an additional thread "to keep the time"