robotology/yarp #1118: Write a doc page containg the list of all the enviromental variables that modify YARP behaviour

traversaro, 2017-03-03:
Such as `YARP_DATA_DIRS`, `YARP_COLORED_OUTPUT`, etc, etc. 

To check which variables are actually used:
* https://github.com/robotology/yarp/search?utf8=%E2%9C%93&q=getenv

drdanz, 2017-03-10:
Also: 

- https://github.com/robotology/yarp/search?q=getEnvironment

Anyway, not all the variables are for general use, see [`VERBOSE_STICA`](https://github.com/robotology/yarp/blob/729eca738ed1b73bf4fb71c8e111892a87cdd654/src/devices/fakeMotionControl/fakeMotionControl.cpp#L354), and [`NONSENSE_ADMIN_DELAY`](https://github.com/robotology/yarp/blob/729eca738ed1b73bf4fb71c8e111892a87cdd654/src/libYARP_OS/src/PortCore.cpp#L2417), other are deprecated or leftovers waiting for cleanup...

drdanz, 2017-03-10:
We should use this issue to take evaluate them and remove whatever is not necessary.

Also note that if I understand it correctly, github search returns only the first match for each file

traversaro, 2017-03-10:
About [`NONSENSE_ADMIN_DELAY`](https://github.com/robotology/yarp/blob/729eca738ed1b73bf4fb71c8e111892a87cdd654/src/libYARP_OS/src/PortCore.cpp#L2417 : 

>  This will be removed soon! 
:D

drdanz, 2017-03-10:
Yeah, as soon as we will have ports!

traversaro, 2017-06-12:
TIL yarp::os::Property expands system enviromental variables (!!): https://github.com/robotology/yarp/blob/729eca738ed1b73bf4fb71c8e111892a87cdd654/src/libYARP_OS/src/Property.cpp#L698

traversaro, 2017-06-12:
## Logger and print configuration
| Enviromental variable | Description |        Related documentation page | 
|:-----------------------:|:------------:|:-------------------------------:|
| `YARP_QUIET`           | ???                | |
| `YARP_VERBOSE`           | ???                | |
| `YARP_LOGGER_STREAM`   | ???                |                      | 
| `YARP_COLORED_OUTPUT`  | If this variable exists and is set to 1, it enables the YARP colored prints. Otherwise disable the colored prints.  |          |
| `YARP_VERBOSE_OUTPUT`  | ???                     |          |
| `YARP_TRACE_ENABLE`    | If this variable exists and is set to 1, it enables the YARP trace prints. Otherwise disable the trace prints.  |          |
| `YARP_DEBUG_ENABLE`    | If this variable exists and is set to 0, it disables the YARP debug prints. Otherwise leaves them enabled. |          |
| `YARP_FORWARD_LOG_ENABLE` | If this variable exists and is set to 1, enables the forwarding of log over ports to be used by the yarplogger. Otherwise disable the forwarding. |          |

## Configuration files
| Enviromental variable | Description |        Related documentation page | 
|:-----------------------:|:------------:|:-------------------------------:|
| `YARP_CONFIG_HOME`     | Location where user config files are stored. | http://www.yarp.it/resource_finder_spec.html |
| `YARP_DATA_HOME`     | Location where user data files are stored. | http://www.yarp.it/resource_finder_spec.html |
| `YARP_CONFIG_DIRS`     | Locations where system administrator data and config files are stored. | http://www.yarp.it/resource_finder_spec.html |
| `YARP_DATA_DIRS`     | Locations where installed data and config files are stored. | http://www.yarp.it/resource_finder_spec.html |
| `YARP_ROBOT_NAME`      | Variable used to refer to the name of the specific robot used in the system, to load its specific configuration files. | http://www.yarp.it/yarp_data_dirs.html |

Note that more platform-specific non-YARP enviromental variables are used when searching for YARP configuration files.
See http://www.yarp.it/resource_finder_spec.html for a full description of the enviromental variables used for finding configuration files.

## UDP Carrier configuration 
| Enviromental variable | Description |        Related documentation page | 
|:-----------------------:|:------------:|:-------------------------------:|
| `YARP_DGRAM_SIZE` | ??? |  | 
| `YARP_MCAST_SIZE` | ??? |  | 
| `YARP_UDP_SIZE` | ??? |  | 
| `YARP_DGRAM_BUFFER_SIZE` | ??? |  | 
| `YARP_DGRAM_RECV_BUFFER_SIZE` | ??? |  | 
| `YARP_DGRAM_SND_BUFFER_SIZE` | ??? |  | 

## ROS configuration
| Enviromental variable | Description |        Related documentation page | 
|:-----------------------:|:------------:|:-------------------------------:|
| `ROS_MASTER_URI`             | If a ROS1 roscore is present in the network, this variable should contain its URI | http://wiki.ros.org/ROS/EnvironmentVariables |
| `YARP_USE_ROS`               | ??? | | 

## Misc configuration
| Enviromental variable | Description |        Related documentation page | 
|:-----------------------:|:------------:|:-------------------------------:|
| `YARP_CLOCK`           | If this variable is set, the default YARP time facilities will read the time from the yarp port whose name is contained in this variable.                | |
| `YARP_RENAME_<portname>`     | ??? |  | 
| `YARP_PORT_PREFIX`           | ??? |  |
| `YARP_STACK_SIZE`           | ???   | |
| `YARP_NAMESPACE`       | ??? |  
| `YARP_IP`           | ??? |  |

## Enviromental variables related to YARP, but not consumed by YARP itself. 

| Enviromental variable | Description | 
|:-----------------------:|:------------:|
| `YARP_DIR`             |  Variable consumed by CMake to find the location of YARP. See https://cmake.org/cmake/help/latest/command/find_package.html for more info  |
| `YARP_ROOT`             | Variable that usually points to the source directory of the YARP repository, used tipically in YARP documentation. |

# Deprecated Enviromental Variables 

| Enviromental variable | Description | 
|:-----------------------:|:------------:|
| `YARP_CONF`               | Legacy variable used to point to the directory where user config files are stored, superseeded by `YARP_CONFIG_HOME` |
| `YARP_POLICY`             |  Legacy variable used to change the behavior of the ResourceFinder. Deprecated since YARP 2.3.65  |

traversaro, 2017-06-12:
The previous comment contains a draft of the proposed documentation. 
I lack appropriate knowledge of several enviromental variables (the one in which the description is `???`) , I would ask some YARP veterans @drdanz @lornat75 @randaz81 to modify the commend to insert the missing descriptions if they know the meaning of such variables (you should be able to modify the comment even if I wrote it, thanks).

drdanz, 2017-06-13:
`YARP_CONF` is no longer accepted in devel, it is just printing a warning, just in case someone was really lazy...

traversaro, 2017-06-13:
`YARP_CONF` removed!

drdanz, 2017-06-13:
Updated `YARP_PORT_PREFIX`

drdanz, 2017-06-13:
Updated `YARP_NAMESPACE`

traversaro, 2017-06-13:
Some info on the env variables used for UDP configuration available in http://www.yarp.it/carrier_config.html .

traversaro, 2017-06-13:
Relevant commits in which some env variables have been introduced: 
* `YARP_DGRAM_BUFFER_SIZE` : https://github.com/robotology/yarp/commit/7d5dedb59109f9ebc01fb08a61600ff48507ea92
* `YARP_DGRAM_RECV_BUFFER_SIZE`/`YARP_DGRAM_SND_BUFFER_SIZE` : https://github.com/robotology/yarp/commit/8c72b50398c402f15990ccd3d7108e1d0669bec4 
The documentation about this env variables can be extracted by the changelog/commit message of those commits.

traversaro, 2017-06-13:
Relevant commit for `YARP_IP` : https://github.com/robotology/yarp/commit/df2e50ebc2b486029d3274b4f020f71f281da784 .

traversaro, 2017-06-13:
Relevant commit for `YARP_STACK_SIZE`: https://github.com/robotology/yarp/commit/b7a63556ad6cda28a2ba5a2f17b20f20ad46c43a

barbalberto, 2017-06-14:
YARP_VERBOSE=<N> set verbosity level of the INTERNAL yarp log system, (not the yDebug... but the internal one which does not use yarp itself, just plain macros and printf)
YARP_QUIET silence the internal log system

So:
YARP_QUIET=0, YARP_VERBOSE=1  --> log message enabled
YARP_QUIET=0, YARP_VERBOSE=4  --> more log message (even though I don't see differences)
YARP_QUIET=0, YARP_VERBOSE=0  --> No log messages
YARP_QUIET=1, YARP_VERBOSE=N  --> No log messages, quiet takes precedence

traversaro, 2017-06-14:
@barbalberto updated, thanks.

traversaro, 2017-06-14:
Relevant commit for `YARP_RENAME_<portname>` : https://github.com/robotology/yarp/commit/f54bd60f054e516428621dbe097d435fe8ba056c

traversaro, 2017-06-14:
`YARP_USE_ROS` :  https://github.com/robotology/yarp/commit/1918fbcbdfbb2cb407e6323aa31c28859d4f410c

drdanz, 2017-06-14:
Thanks @traversaro for taking care of this