robotology/yarp #1121: Compiling yarp devel with RTF enabled configure fines even if the right version of RTF  is not found

traversaro, 2017-03-07:
`cmake -DYARP_COMPILE_RTF_ADDONS:BOOL=ON ..` works fine even if the wrong version of RTF is found, and then the compilation fails with the following error:
~~~
CMake Error at src/libYARP_rtf/CMakeLists.txt:27 (add_library):
  Target "YARP_rtf" links to target "RTF::RTF" but the target was not found.
  Perhaps a find_package() call is missing for an IMPORTED target, or an
  ALIAS target is missing?

CMake Error at src/rtf-plugins/fixture-managers/yarpmanager/CMakeLists.txt:17 (add_library):
  Target "rtf_fixturemanager_yarpmanager" links to target "RTF::RTF" but the
  target was not found.  Perhaps a find_package() call is missing for an
  IMPORTED target, or an ALIAS target is missing?

CMake Error at src/rtf-plugins/fixture-managers/yarpplugin/CMakeLists.txt:18 (add_library):
  Target "rtf_fixturemanager_yarpplugin" links to target "RTF::RTF" but the
  target was not found.  Perhaps a find_package() call is missing for an
  IMPORTED target, or an ALIAS target is missing?

CMake Error at src/libYARP_rtf/CMakeLists.txt:27 (add_library):
  Target "YARP_rtf" links to target "RTF::RTF" but the target was not found.
  Perhaps a find_package() call is missing for an IMPORTED target, or an
  ALIAS target is missing?
~~~

JamesKitching, 2017-03-17:
I am having the same issue after installing Mathlab and trying to run the iCub schools seminar build https://github.com/vvv-school/vvv17/wiki/Before-arriving-at-VVV.  I am assuming I am okay to attempt to do this as the projects are all open source.  If I could get this working I would be keen to setup an open source project of my own in the area.  Thanks

traversaro, 2017-03-17:
Hi @JamesKitching , 
   this error could arise if you installed yarp from the `devel` branch, while rtf (https://github.com/robotology/robot-testing/tree/devel) from the `master` branch. If you install both from either the `master` or the `devel` branch, everything should work fine. See https://github.com/robotology/yarp/pull/1097 for more info.

JamesKitching, 2017-03-17:
Thanks for the heads up.

drdanz, 2017-03-18:
Actually, we already check that RTF is the devel version, see https://github.com/robotology/yarp/blob/devel/cmake/YarpFindDependencies.cmake#L332
So if you get this error there might be a bug here... I'll check on Monday...

drdanz, 2017-03-20:
Nope, I cannot reproduce it... @traversaro can you tell me something more about the setup where you had this issue?

JamesKitching, 2017-03-20:
I was running the VMWARE image dsecribed on https://github.com/vvv-school/vvv17/wiki/Before-arriving-at-VVV. See "Download our Virtual Machine We have made available an Ubuntu 16.04 VM based on VMWare.".  After installing Math Lab and executing the build command "sudo ~/Downloads/matlab_R2106b_glnxa64/install"  I got the error.  I did think to try to do a build from scratch locally after the dev / master comment (I assumed the build did an automated background get that broke it) but then I ran into the problem of having to copy my Math Lab trial out of VMWare.  It was at this point that I gave up and started looking at KnowRob instead - which now comes with an Android version.  I might be tempted back at some point as getting a paid copy of Simulink would be nice (I use to work on a competitor product).

drdanz, 2017-03-21:
I think the issue might be in the config files in RTF, that apparently do not fail when the requested version is not correct.
CC @apaikan