robotology/yarp #1122: Problems displaying some graphics items

S-Dafarra, 2017-03-08:
I recently installed from scratch Ubuntu 16.04, YARP and iCub software. Testing the installation I noticed that the motorgui looks weird, much "bigger" than expected.
[image]
[image]

 The yarpmanager interface instead is ok, while the yarpdataplayer seems to have the same problem for some icons
[image]

It may be related to #837 and #1038.

drdanz, 2017-03-10:
Do you have a normal display or a high DPI display?

S-Dafarra, 2017-03-10:
A normal full HD laptop display. If it may help, I have an NVIDIA GeForce 830M graphic card using the open-source driver X.Org X server.

Nicogene, 2017-05-08:
Hi @S-Dafarra !

Which version of `Qt` do you have ?

S-Dafarra, 2017-05-08:
By running ```qmake --version``` I get 4.8.7, but I think to have both ``qt4`` and ``qt5``.

S-Dafarra, 2017-05-08:
Actually I tried now to install ``qt5-default`` and by performing again ``qmake --version`` it returns  5.5.1, but after recompiling ``icub-main`` I get the same weird visualization.

Nicogene, 2017-05-08:
The guis are in `yarp` so if you compile only `icub-main` you can't see if qt5 enabled as default does the trick :smile: .
#1200 introduce some changes that could fix this issue.
My advice is to update your `yarp devel` pulling from `origin\devel`, compile and see if the problem persist.
I have your setup (except for the nvidia gpu) and I see the guis normally.
Let me know if the last commit fixed this issue.

S-Dafarra, 2017-05-08:
Ops :sweat_smile:, newbie error. Now is working fine! Thanks

closed 2017-05-08 by Nicogene