robotology/yarp #1130: problem with installing YARP

azadm, 2017-03-14:
I get this error after ``make`` command:

````
Scanning dependencies of target YARP_priv_qcustomplot_automoc
[  0%] Automatic moc for target YARP_priv_qcustomplot
Generating moc source YARP_priv_qcustomplot_automoc.dir/moc_qcustomplot_JLIHGTVE57VUXS.cpp
Generating moc compilation YARP_priv_qcustomplot_automoc.cpp
[  0%] Built target YARP_priv_qcustomplot_automoc
Scanning dependencies of target YARP_priv_qcustomplot
[  0%] Building CXX object extern/qcustomplot/CMakeFiles/YARP_priv_qcustomplot.dir/qcustomplot/qcustomplot.cpp.o
[  1%] Building CXX object extern/qcustomplot/CMakeFiles/YARP_priv_qcustomplot.dir/YARP_priv_qcustomplot_automoc.cpp.o
[  1%] Linking CXX static library ../../lib/libYARP_priv_qcustomplot.a
[  1%] Built target YARP_priv_qcustomplot
Scanning dependencies of target YARP_priv_xmlrpcpp
[  1%] Building CXX object extern/xmlrpcpp/CMakeFiles/YARP_priv_xmlrpcpp.dir/xmlrpcpp/src/base64.cpp.o
[  1%] Building CXX object extern/xmlrpcpp/CMakeFiles/YARP_priv_xmlrpcpp.dir/xmlrpcpp/src/XmlRpcClient.cpp.o
[  1%] Building CXX object extern/xmlrpcpp/CMakeFiles/YARP_priv_xmlrpcpp.dir/xmlrpcpp/src/XmlRpcServerConnection.cpp.o
[  2%] Building CXX object extern/xmlrpcpp/CMakeFiles/YARP_priv_xmlrpcpp.dir/xmlrpcpp/src/XmlRpcServer.cpp.o
[  2%] Building CXX object extern/xmlrpcpp/CMakeFiles/YARP_priv_xmlrpcpp.dir/xmlrpcpp/src/XmlRpcServerMethod.cpp.o
[  2%] Building CXX object extern/xmlrpcpp/CMakeFiles/YARP_priv_xmlrpcpp.dir/xmlrpcpp/src/XmlRpcSource.cpp.o
[  2%] Building CXX object extern/xmlrpcpp/CMakeFiles/YARP_priv_xmlrpcpp.dir/xmlrpcpp/src/XmlRpcUtil.cpp.o
[  2%] Building CXX object extern/xmlrpcpp/CMakeFiles/YARP_priv_xmlrpcpp.dir/xmlrpcpp/src/XmlRpcValue.cpp.o
[  3%] Linking CXX static library ../../lib/libYARP_priv_xmlrpcpp.a
[  3%] Built target YARP_priv_xmlrpcpp
Scanning dependencies of target YARP_OS
[  3%] Building CXX object src/libYARP_OS/CMakeFiles/YARP_OS.dir/src/AbstractCarrier.cpp.o
In file included from /Users/azadm/Applications/yarp/src/libYARP_OS/src/AbstractCarrier.cpp:9:
In file included from /Users/azadm/Applications/yarp/src/libYARP_OS/include/yarp/os/impl/Logger.h:16:
In file included from /usr/local/include/ace/Log_Msg.h:22:
In file included from /usr/local/include/ace/Default_Constants.h:28:
In file included from /usr/local/include/ace/os_include/os_limits.h:25:
In file included from /usr/local/include/ace/os_include/os_unistd.h:25:
/usr/local/include/ace/os_include/sys/os_types.h:49:16: error: typedef redefinition with different types
      ('int' vs 'enum clockid_t')
   typedef int clockid_t;
               ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk/usr/include/time.h:171:3: note: 
      previous definition is here
} clockid_t;
  ^
In file included from /Users/azadm/Applications/yarp/src/libYARP_OS/src/AbstractCarrier.cpp:9:
In file included from /Users/azadm/Applications/yarp/src/libYARP_OS/include/yarp/os/impl/Logger.h:16:
In file included from /usr/local/include/ace/Log_Msg.h:746:
In file included from /usr/local/include/ace/Log_Msg.inl:2:
In file included from /usr/local/include/ace/OS_NS_string.h:26:
In file included from /usr/local/include/ace/Basic_Types.h:54:
In file included from /usr/local/include/ace/os_include/os_stdlib.h:26:
In file included from /usr/local/include/ace/os_include/sys/os_wait.h:26:
In file included from /usr/local/include/ace/os_include/sys/os_resource.h:25:
/usr/local/include/ace/os_include/sys/os_time.h:32:11: warning: 'CLOCK_REALTIME' macro redefined
      [-Wmacro-redefined]
#  define CLOCK_REALTIME 0
          ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk/usr/include/time.h:154:9: note: 
      previous definition is here
#define CLOCK_REALTIME _CLOCK_REALTIME
        ^
In file included from /Users/azadm/Applications/yarp/src/libYARP_OS/src/AbstractCarrier.cpp:9:
In file included from /Users/azadm/Applications/yarp/src/libYARP_OS/include/yarp/os/impl/Logger.h:16:
In file included from /usr/local/include/ace/Log_Msg.h:746:
In file included from /usr/local/include/ace/Log_Msg.inl:2:
In file included from /usr/local/include/ace/OS_NS_string.h:26:
In file included from /usr/local/include/ace/Basic_Types.h:54:
In file included from /usr/local/include/ace/os_include/os_stdlib.h:26:
In file included from /usr/local/include/ace/os_include/sys/os_wait.h:26:
In file included from /usr/local/include/ace/os_include/sys/os_resource.h:25:
/usr/local/include/ace/os_include/sys/os_time.h:36:11: warning: 'CLOCK_MONOTONIC' macro redefined
      [-Wmacro-redefined]
#  define CLOCK_MONOTONIC 1
          ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk/usr/include/time.h:156:9: note: 
      previous definition is here
#define CLOCK_MONOTONIC _CLOCK_MONOTONIC
        ^
2 warnings and 1 error generated.
make[2]: *** [src/libYARP_OS/CMakeFiles/YARP_OS.dir/src/AbstractCarrier.cpp.o] Error 1
make[1]: *** [src/libYARP_OS/CMakeFiles/YARP_OS.dir/all] Error 2
make: *** [all] Error 2

````

I am using macOS Sierra version 10.12.3

francesco-romano, 2017-03-14:
Hi @azadm 
Do you have the most recent version of ace installed?

azadm, 2017-03-14:
yes I guess so:
````
dynamic200-113:~ azadm$ brew install ace
Warning: ace-6.4.1 already installed
````

francesco-romano, 2017-03-14:
Yes, it seems so.
This is strange as that issue has been solved long ago, see [here](https://github.com/Homebrew/homebrew-core/issues/3639)

azadm, 2017-03-14:
I managed to install ``yarp`` using ``homebrew`` but still having problem while trying to install from the source.

claudiofantacci, 2017-03-19:
Are there any updates on this issue?

azadm, 2017-03-20:
I managed to install both ``yarp`` and ``icub-main`` from source. I don't know why but I had to remove everything and install from scratch...maybe due to the upgrade to Sierra.

closed 2017-03-20 by claudiofantacci