robotology/yarp #1132: yarpidl_rosmsg fails on AppVeyor

francesco-romano, 2017-03-16:
But successful on Windows build

See log:

```
-- thrift code for thrift/XsensFrame.thrift => C:/dev/human-dynamics-estimation/build/xsens-tf-bridge/autogenerated/data
-- thrift code for thrift/XsensDriverService.thrift => C:/dev/human-dynamics-estimation/build/xsens-tf-bridge/autogenerated/services
-- rosmsg code for msgs/String.msg => C:/dev/human-dynamics-estimation/build/xsens-tf-bridge/autogenerated/msgs
CMake Error at C:/Program Files (x86)/robotology/yarp-2.3.68/share/yarp/cmake/YarpIDL.cmake:124 (message):
  yarpidl_rosmsg (C:/Program Files
  (x86)/robotology/yarp-2.3.68/bin/yarpidl_rosmsg.exe) failed, aborting.
Call Stack (most recent call first):
  xsens-tf-bridge/CMakeLists.txt:15 (yarp_idl_to_dir)
-- Configuring incomplete, errors occurred!

```

This is the AppVeyor config file
```
version: 1.0.{build}

clone_folder: c:\dev\human-dynamics-estimation

environment:
  YARP_DIR:          c:/Program Files (x86)/robotology/yarp-2.3.68
  EIGEN3_ROOT:       c:/Program Files (x86)/robotology/eigen-3.3.0
  
install:
  # to install unzipper do: "cinst 7zip.commandline -x86"
  # the format is: "7z x c:\path-to-archive\archive.zip -oc:\path-where-unzip -y > nul"
  - ps: >-
      $dwntool = New-Object System.Net.WebClient

      $Stoploop = $false

      do {
          try {
            $dwntool.DownloadFile('https://github.com/robotology/yarp/releases/download/v2.3.68/yarp_2.3.68_v12_x86_1.exe','c:\dev\yarp.exe')
            $dwntool.DownloadFile('https://github.com/robotology/ycm/releases/download/v0.2.2/YCM-0.2.2-win32.msi','c:\dev\ycm.msi')
            $Stoploop = $true
          }
          catch {
            Start-Sleep -Seconds 15
          }
      }
      While ($Stoploop -eq $false)

  - cmd: start "" /WAIT "c:\dev\yarp.exe" /S
  - cmd: timeout 10
  - cmd: msiexec.exe /i "c:\dev\ycm.msi" /quiet
  - cmd: timeout 10
  - cmd: dir "c:\Program Files (x86)\robotology"

  - cmd: git clone https://github.com/robotology/idyntree.git c:\dev\idyntree
  - cmd: cd c:\dev\idyntree
  - cmd: md build
  - cmd: cd build
  - cmd: cmake -G"Visual Studio 12" -DIDYNTREE_USES_KDL=OFF -DIDYNTREE_USES_YARP=ON -DIDYNTREE_USES_ICUB_MAIN=OFF  ..
  - cmd: msbuild /m /p:Configuration=Release /p:Platform="Win32" INSTALL.vcxproj

build:

build_script:
  - cd c:\dev\human-dynamics-estimation\
  - md build
  - cd build
  - cmake -G"Visual Studio 12" ..
  - msbuild /m /p:Configuration=Release /p:Platform="Win32" HumanDynamicsEstimation.sln
```

drdanz, 2017-03-24:
Can you add a link to this `xsens-tf-bridge/CMakeLists.txt` file?

francesco-romano, 2017-03-30:
@drdanz https://github.com/robotology-playground/human-dynamics-estimation/blob/master/human-viz-bridge/CMakeLists.txt

But it happens also on other repositories (not on the idl_ros but on the thrift generation)

drdanz, 2018-03-12:
This should be fixed, whatever the problem was. IDL tests are currently running on appveyor

closed 2018-03-12 by drdanz