robotology/yarp #1135: yarp::os::Subscriber segfault when using callback

randaz81, 2017-03-21:
The following code leads to a segfault (the first time data are received)
```
m_dataReader->new DataReaderClass;
m_dataReader->useCallback();
m_dataReader->topic("/ros_topic");
```
while the following is ok:
```
m_dataReader->new DataReaderClass;
m_dataReader->topic("/ros_topic");
m_dataReader->useCallback();
```
Being DataReaderClass defined as:
```
class DataReaderClass: public yarp::os::Subscriber<ros_msg_type>
```

The issue could be related to the fact that `topic()` automatically creates a connection? (The connection is typically created later when using a standard buffered port with callback, instead).

randaz81, 2020-07-24:
I can confirm this bug is still present in yarp 3.3 [master branch]

Additional note: if I use a `yarp::os::Port `class, I am able to call the `useCallback()` method before or after the `.open()` method with no segfault. So this bug seems to be somehow specific for the `yarp::os::Subscriber` class.

ste93, 2021-11-16:
[image]
Same error happened today, This is the backtrace generated by gdb. It seems that the topic is not completely started when the first callback is called