robotology/yarp #1139: Cleanup Yarp Motor Interfaces documentation

traversaro, 2017-03-23:
We have several points of introduction to the Yarp Motor Interfaces. 
I list the one that I found, and my proposal on what to do with them: 

|    Material     |    Proposed action                                           | 
|:------------:|:-------------------------------------------------:|
| https://github.com/robotology/yarp/blob/master/doc/yarp_motor_control.dox |  **KEEP IT**, Make this the main entry point for documentation of Yarp Motor Interfaces, cleanup of old and outdated material and add links to any relevant material. |
| http://wiki.icub.org/wiki/Motor_control | **DELETE IT** ,  Superseded by http://wiki.icub.org/brain/icub_motor_control_tutorial.html |
| http://wiki.icub.org/wiki/YARP_Motor_Control | Merge its content with https://github.com/robotology/yarp/blob/master/doc/yarp_motor_control.dox, and **DELETE IT** |
| http://www.yarp.it/group__dev__iface__motor.html | Group containing all the motor control interfaces. **KEEP IT** and link it to the main yarp_motor_control.dox page |
| http://wiki.icub.org/wiki/Force_Control#Available_control_modes | **DELETE IT**, and move the interesting part to some doxygen documentation | 
| https://github.com/robotology/icub-tutorials/blob/master/doc/motor_control_tutorial.dox | This is tough. It is probably the best tutorial on motor control, so it would make sense to have something similarly in YARP rather then icub-tutorials. The possibility are to move it in YARP, and have some similar tutorial in YARP, leaving this one in `icub-tutorials`. **I DO NOT KNOW** | 
| http://wiki.icub.org/wiki/Control_Modes | The PDF describing the documentation of the control modes is a great work, but it is definitely more YARP-related then just being related to iCub, but I don't know where we could put it. Committing the PDF to the repo is probably undesirable, right? **I DO NOT KNOW** |

drdanz, 2017-03-23:
+1 for removing all YARP-only documentation from iCub wiki and move it in YARP

> |    Material     |    Proposed action                                           | 
> |:------------:|:-------------------------------------------------:|
> | http://wiki.icub.org/wiki/Control_Modes | The PDF describing the documentation of the control modes is a great work, but it is definitely more YARP-related then just being related to iCub, but I don't know where we could put it. Committing the PDF to the repo is probably undesirable, right? **I DO NOT KNOW** |  

The PDF is a very good work, but I'd like to see it in YARP documentation and PDF is not the right choice. It is not easy to search inside it, and this makes it difficult to keep it updated. For example, I think that @randaz81 is working on a couple of new control modes, but I don't know if he will remember to update this doc... 😉 

Perhaps we can copy-paste it in a markdown file and perhaps use either doxygen or some other specific tool (for example [this](https://www.npmjs.com/package/markdown-pdf)) to re-generate the PDF? (Assuming that we actually need it, having them in yarp documentation looks good enough to me)

randaz81, 2017-05-18:
I'm currently working on http://wiki.icub.org/wiki/Control_Modes