robotology/yarp #1140: test_grabber description has weird default values and lacks descriptions of the parameters

drdanz, 2017-03-24:
The default values here are sequential 4.0, 5.0, 6.0... this sounds very weird to me, are you sure that these are the correct parameters?
```c++
    intrinsic.put("focalLengthX",config.check("focalLengthX",Value(4.0),"").asDouble());
    intrinsic.put("focalLengthY",config.check("focalLengthY",Value(5.0),"").asDouble());
    intrinsic.put("principalPointX",config.check("principalPointX",Value(6.0),"").asDouble());
    intrinsic.put("principalPointY",config.check("principalPointY",Value(7.0),"").asDouble());
    intrinsic.put("retificationMatrix",config.check("retificationMatrix",*retM,""));
    intrinsic.put("distortionModel",config.check("distortionModel",Value("FishEye"),"").asString());
    intrinsic.put("k1",config.check("k1",Value(8.0),"").asDouble());
    intrinsic.put("k2",config.check("k2",Value(9.0),"").asDouble());
    intrinsic.put("k3",config.check("k3",Value(10.0),"").asDouble());
    intrinsic.put("t1",config.check("t1",Value(11.0),"").asDouble());
    intrinsic.put("t2",config.check("t2",Value(12.0),"").asDouble());
```

Also the documentation for these parameters is completely missing (it should be in the `""` in order to have it printed on the console when you run `yarpdev --verbose`

Nicogene, 2017-03-24:
Hi @drdanz,
they are not the "correct" parameters, I used these numbers only to be sure in `TestFrameGrabberTest` that all the parameters are correctly set in the `intrinsic` Property and can be read through `rgbParams->getRgbIntrinsicParam(intrinsic)`.

The correct default  values for these parameters are almost all `0.0`.
The default values I chose doesn't impact the functionality of the `TestFrameGrabber`, because it is simply a fake device.
If you prefer I can change to more appropriate default values.
For the documentation I forgot to add it  :sweat_smile:

drdanz, 2017-03-24:
Wait, I think there is some confusion here... you mean that the frame grabber is _not_ able to perform an actual distortion, it is just setting these as intrinsic parameters? Please update the documentation of the parameters, explaining this...

Nicogene, 2017-03-24:
Yeah, it just set the intrinsic parameters for testing the "new" interface `IRGBVisualParams` we add to `TestFrameGrabber`.
Ok, I'm on it :wink:

Nicogene, 2017-04-13:
Fixed by #1171

closed 2017-05-03 by Nicogene