robotology/yarp #1149: Compiling yarp with support for python 3.5

pedrombmachado, 2017-03-28:
Dear all,
  I was trying to compile the YARP with support for python 3.5 and was unable to do that. One of the issues was related with having half of the links for python 2.7 and another half for python 3.5.
check here:
$ grep -R "PYTHON" ../ | grep PATH
../bindings/tests/python/run.sh:    export PYTHONPATH=$YARP_DIR:$PYTHONPATH
../bindings/tests/python/run.sh:    export PYTHONPATH=$PWD:$PYTHONPATH
../bindings/CMakeLists.txt:  include_directories(SYSTEM ${PYTHON_INCLUDE_PATH})
../bindings/CMakeLists.txt:    get_filename_component(PYTHON_DIR ${PYTHON_LIBRARY} PATH)
../bindings/CMakeLists.txt:  set(CMAKE_INSTALL_PYTHONDIR ${_CMAKE_INSTALL_PYTHONDIR} CACHE PATH "python bindings (${_CMAKE_INSTALL_PYTHONDIR})")
../bindings/CMakeLists.txt:    file(RELATIVE_PATH _rel_path "${CMAKE_INSTALL_FULL_PYTHONDIR}" "${CMAKE_INSTALL_FULL_LIBDIR}")
../doc/yarp_swig.dox:For Python, you can append the path of the bindings build directory to the PYTHONPATH environment variable, like this:
../doc/yarp_swig.dox:export PYTHONPATH=$PYTHONPATH:/path/to/bindings/build
../build/CMakeCache.txt:CMAKE_INSTALL_PYTHONDIR:PATH=lib/python2.7/dist-packages
../build/CMakeCache.txt:PYTHON_EXECUTABLE:FILEPATH=/usr/bin/python
../build/CMakeCache.txt:PYTHON_INCLUDE_DIR:PATH=/usr/include/python2.7
../build/CMakeCache.txt:PYTHON_LIBRARY:FILEPATH=/usr/lib/x86_64-linux-gnu/libpython2.7.so
../build/CMakeCache.txt:PYTHON_LIBRARY_DEBUG:FILEPATH=PYTHON_LIBRARY_DEBUG-NOTFOUND

The other issue was related with having python installing the yarp.py and _yarp.so
-- Installing: /usr/local/lib/python3/dist-packages/_yarp.so
-- Installing: /usr/local/lib/python3/dist-packages/yarp.py

$ yarp version
YARP version 2.3.68+205-20170303.1+git1993603

$ cmake --version
cmake version 3.5.1

$ uname -a
Linux icubsrv 4.4.0-66-generic #87-Ubuntu SMP Fri Mar 3 15:29:05 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux

Tobias-Fischer, 2017-03-31:
Hi, did you try setting the `YARP_USE_PYTHON_VERSION` cmake variable?

pedrombmachado, 2017-03-31:
Hi @Tobias-Fischer. yes I did. I entered 3.5

drdanz, 2017-05-10:
@pedrombmachado have you tried cleaning the cache and/or removing the build directory?

Also ensure that the `YARP_USE_PYTHON_VERSION` variable is set as soon as possible (maybe run `cmake -DYARP_USE_PYTHON_VERSION=3.5`) perhaps not all the variables are replaced when FindPython is executed the second time.

gsaponaro, 2018-01-03:
I stumbled upon this issue, too. The suggestion by @drdanz worked, in my case I enforced the following variables as soon as possible from the build directory:

`cmake -DYARP_COMPILE_BINDINGS:BOOL=ON -DCREATE_PYTHON:BOOL=ON -DYARP_USE_PYTHON_VERSION=3.5 -DPYTHON_INCLUDE_DIR=/usr/include/python3.5m -DPYTHON_LIBRARY=/usr/lib/x86_64-linux-gnu/libpython3.5m.so.1 ..`

drdanz, 2018-01-09:
@gsaponaro Thanks for testing this.
I'm closing this issue, @pedrombmachado please re-open it if you still have problems with this.

closed 2018-01-09 by drdanz