robotology/yarp #1154: yarpmotorgui does not read parameters from command line

valeriosperati, 2017-03-30:
Hi, I'm trying to pass parameters to yarpmotorgui from command line
`yarpmotorgui --name icubSim`

following the instructions shown here
[http://www.yarp.it/yarpmotorgui.html](url)

Nevertheless the gui opens always with the default robot name 'icub'

Test on Debian 8.7 Jessie.
Thanks

claudiofantacci, 2017-03-30:
I think the documentation needs to be updated.

The command is `yarpmotorgui --robot icubSim`.

What is the YARP branch you are currently working on?
Please also type `yarp version` in the terminal and tell us the output.

valeriosperati, 2017-03-31:
mmm no, neither --robot works. The motorgui continues to open with default name icub.
yarp version:
 `YARP version 2.3.68+218-20170316.1+git3050aeb`

git status (run in YARP_DIR):
`On branch master`

claudiofantacci, 2017-03-31:
@valeriosperati I'm using `YARP version 2.3.68+218-20170316.1+git3050aeb11`, please do a `git pull --rebase` to synch to the latest version.

Everything is working fine on my side.
Using `yarpmotorgui` with no argument prompts the following dialog
[image]
from which you can change the default name manually.
You can however use `yarpmotorgui --name icubSim` and the dialog modifies as follows
[image]

valeriosperati, 2017-04-03:
mmm no luck. I did perform 
`git pull --rebase`
and recompiled.

launching 
`yarpmotorgui --name icubSim`
continues to open icub robot.
This is the console output, maybe some info is here:

```
yarpmotorgui --name icubSim
||| configuring
||| default config file specified as yarpmotorgui.ini
||| checking [/home/valeriosperati/Desktop/yarpmotorgui.ini] (pwd)
||| checking [/home/valeriosperati/.config/yarp/robots/default] (robot YARP_CONFIG_HOME)
||| checking [/home/valeriosperati/.local/share/yarp/robots/default] (robot YARP_DATA_HOME)
||| checking [/etc/xdg/yarp/robots/default] (robot YARP_CONFIG_DIRS)
||| checking [/home/valeriosperati/Desktop/VALERIO_SPERATI/SORGENTI/yarp/build/share/yarp/robots/default] (robot YARP_DATA_DIRS)
||| checking [/home/valeriosperati/Desktop/VALERIO_SPERATI/SORGENTI/ICUB/icub-main/build/share/iCub/robots/default] (robot YARP_DATA_DIRS)
||| checking [/home/valeriosperati/Desktop/VALERIO_SPERATI/SORGENTI/ICUB/icub-contrib-common/build/share/ICUBcontrib/robots/default] (robot YARP_DATA_DIRS)
||| checking [/home/valeriosperati/Desktop/VALERIO_SPERATI/SORGENTI/ICUB/icub-main/app/robots/default] (robot YARP_DATA_DIRS)
||| checking [/home/valeriosperati/Desktop/VALERIO_SPERATI/SORGENTI/yarp/build/share/yarp/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/home/valeriosperati/Desktop/VALERIO_SPERATI/SORGENTI/ICUB/icub-main/build/share/iCub/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/home/valeriosperati/Desktop/VALERIO_SPERATI/SORGENTI/ICUB/icub-contrib-common/build/share/ICUBcontrib/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/home/valeriosperati/Desktop/VALERIO_SPERATI/SORGENTI/ICUB/icub-main/app/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/home/valeriosperati/.config/yarp/yarpmotorgui.ini] (YARP_CONFIG_HOME)
||| checking [/home/valeriosperati/.local/share/yarp/yarpmotorgui.ini] (YARP_DATA_HOME)
||| checking [/etc/xdg/yarp/yarpmotorgui.ini] (YARP_CONFIG_DIRS)
||| checking [/home/valeriosperati/Desktop/VALERIO_SPERATI/SORGENTI/yarp/build/share/yarp/yarpmotorgui.ini] (YARP_DATA_DIRS)
||| checking [/home/valeriosperati/Desktop/VALERIO_SPERATI/SORGENTI/ICUB/icub-main/build/share/iCub/yarpmotorgui.ini] (YARP_DATA_DIRS)
||| checking [/home/valeriosperati/Desktop/VALERIO_SPERATI/SORGENTI/ICUB/icub-contrib-common/build/share/ICUBcontrib/yarpmotorgui.ini] (YARP_DATA_DIRS)
||| checking [/home/valeriosperati/Desktop/VALERIO_SPERATI/SORGENTI/ICUB/icub-main/app/yarpmotorgui.ini] (YARP_DATA_DIRS)
||| did not find yarpmotorgui.ini
yarp: cannot read from yarpmotorgui.ini
[DEBUG]Robot name: icub
Setting default parts.
[DEBUG]Appending head
[DEBUG]Appending torso
[DEBUG]Appending left_arm
[DEBUG]Appending right_arm
[DEBUG]Appending left_leg
[DEBUG]Appending right_leg
```

Note that I did not installed yarp.
YARP_DIR points to my yarp build folder; 
Thanks

claudiofantacci, 2017-04-03:
This is utterly strange. It works just fine to me.

What is the output of `which yarpmotorgui `?
What is the folder path of your build directory?

valeriosperati, 2017-04-03:
here it is:
`which yarpmotorgui`
`/home/valeriosperati/Desktop/VALERIO_SPERATI/SORGENTI/yarp/build/bin/yarpmotorgui`

this is YARP_DIR
`echo $YARP_DIR`
`/home/valeriosperati/Desktop/VALERIO_SPERATI/SORGENTI/yarp/build`

I tested the motorgui command  command on other two desktop (both with debian): command line arguments are skipped there too.

claudiofantacci, 2017-04-03:
I have tested `yarpmotorgui` with arguments on macOS, Debian, Ubuntu and Windows machines and works fine.

Could you please provide us the output of `env` command.

valeriosperati, 2017-04-03:
yep! env:

```
XDG_VTNR=7
FRITZING_DIR=/home/valeriosperati/Desktop/VALERIO_SPERATI/SORGENTI/fritzing-0.9.3b.linux.AMD64
SSH_AGENT_PID=1293
XDG_SESSION_ID=1
XDG_GREETER_DATA_DIR=/var/lib/lightdm/data/valeriosperati
GPG_AGENT_INFO=/tmp/gpg-ob5pux/S.gpg-agent:1314:1
GLADE_PIXMAP_PATH=:
TERM=xterm
SHELL=/bin/bash
XDG_MENU_PREFIX=xfce-
YARP_DIR=/home/valeriosperati/Desktop/VALERIO_SPERATI/SORGENTI/yarp/build
GAZEBO_MODEL_PATH=/home/valeriosperati/Desktop/VALERIO_SPERATI/SORGENTI/Gazebo_fr/icub-gazebo
WINDOWID=67115294
USER=valeriosperati
LD_LIBRARY_PATH=/home/valeriosperati/Desktop/VALERIO_SPERATI/SORGENTI/yarp/build/lib
LS_COLORS=rs=0:di=01;34:ln=01;36:mh=00:pi=40;33:so=01;35:do=01;35:bd=40;33;01:cd=40;33;01:or=40;31;01:su=37;41:sg=30;43:ca=30;41:tw=30;42:ow=34;42:st=37;44:ex=01;32:*.tar=01;31:*.tgz=01;31:*.arc=01;31:*.arj=01;31:*.taz=01;31:*.lha=01;31:*.lz4=01;31:*.lzh=01;31:*.lzma=01;31:*.tlz=01;31:*.txz=01;31:*.tzo=01;31:*.t7z=01;31:*.zip=01;31:*.z=01;31:*.Z=01;31:*.dz=01;31:*.gz=01;31:*.lrz=01;31:*.lz=01;31:*.lzo=01;31:*.xz=01;31:*.bz2=01;31:*.bz=01;31:*.tbz=01;31:*.tbz2=01;31:*.tz=01;31:*.deb=01;31:*.rpm=01;31:*.jar=01;31:*.war=01;31:*.ear=01;31:*.sar=01;31:*.rar=01;31:*.alz=01;31:*.ace=01;31:*.zoo=01;31:*.cpio=01;31:*.7z=01;31:*.rz=01;31:*.cab=01;31:*.jpg=01;35:*.jpeg=01;35:*.gif=01;35:*.bmp=01;35:*.pbm=01;35:*.pgm=01;35:*.ppm=01;35:*.tga=01;35:*.xbm=01;35:*.xpm=01;35:*.tif=01;35:*.tiff=01;35:*.png=01;35:*.svg=01;35:*.svgz=01;35:*.mng=01;35:*.pcx=01;35:*.mov=01;35:*.mpg=01;35:*.mpeg=01;35:*.m2v=01;35:*.mkv=01;35:*.webm=01;35:*.ogm=01;35:*.mp4=01;35:*.m4v=01;35:*.mp4v=01;35:*.vob=01;35:*.qt=01;35:*.nuv=01;35:*.wmv=01;35:*.asf=01;35:*.rm=01;35:*.rmvb=01;35:*.flc=01;35:*.avi=01;35:*.fli=01;35:*.flv=01;35:*.gl=01;35:*.dl=01;35:*.xcf=01;35:*.xwd=01;35:*.yuv=01;35:*.cgm=01;35:*.emf=01;35:*.axv=01;35:*.anx=01;35:*.ogv=01;35:*.ogx=01;35:*.aac=00;36:*.au=00;36:*.flac=00;36:*.m4a=00;36:*.mid=00;36:*.midi=00;36:*.mka=00;36:*.mp3=00;36:*.mpc=00;36:*.ogg=00;36:*.ra=00;36:*.wav=00;36:*.axa=00;36:*.oga=00;36:*.spx=00;36:*.xspf=00;36:
XDG_SESSION_PATH=/org/freedesktop/DisplayManager/Session0
GLADE_MODULE_PATH=:
XDG_SEAT_PATH=/org/freedesktop/DisplayManager/Seat0
SSH_AUTH_SOCK=/tmp/ssh-cxSaSOGADg7j/agent.1268
SESSION_MANAGER=local/debian:@/tmp/.ICE-unix/1307,unix/debian:/tmp/.ICE-unix/1307
XDG_CONFIG_DIRS=/etc/xdg
ICUB_DIR=/home/valeriosperati/Desktop/VALERIO_SPERATI/SORGENTI/ICUB/icub-main
PATH=/usr/local/bin:/usr/bin:/bin:/usr/local/games:/usr/games:/home/valeriosperati/Desktop/VALERIO_SPERATI/SORGENTI/yarp/build/bin:/home/valeriosperati/Desktop/VALERIO_SPERATI/SORGENTI/ICUB/icub-main/build/bin:/home/valeriosperati/Desktop/VALERIO_SPERATI/SORGENTI/ICUB/icub-contrib-common/build/bin:/home/valeriosperati/Android/Sdk/platform-tools:/home/valeriosperati/Desktop/VALERIO_SPERATI/SORGENTI/processing-3.2.3:/home/valeriosperati/Desktop/VALERIO_SPERATI/SORGENTI/fritzing-0.9.3b.linux.AMD64
DESKTOP_SESSION=lightdm-xsession
OPENCV_TEST_DATA_PATH=/home/valeriosperati/Desktop/VALERIO_SPERATI/SORGENTI/opencv_extra-master/testdata
PWD=/home/valeriosperati
LANG=en_GB.utf8
GAZEBO_DIR=/home/valeriosperati/Desktop/VALERIO_SPERATI/SORGENTI/Gazebo_fr/icub-gazebo
OpenCV_DIR=/home/valeriosperati/Desktop/VALERIO_SPERATI/SORGENTI/opencv-2.4.13/build
ICUBcontrib_DIR=/home/valeriosperati/Desktop/VALERIO_SPERATI/SORGENTI/ICUB/icub-contrib-common/build
PROCESSING_DIR=/home/valeriosperati/Desktop/VALERIO_SPERATI/SORGENTI/processing-3.2.3
YARP_DATA_DIRS=/home/valeriosperati/Desktop/VALERIO_SPERATI/SORGENTI/yarp/build/share/yarp:/home/valeriosperati/Desktop/VALERIO_SPERATI/SORGENTI/ICUB/icub-main/build/share/iCub:/home/valeriosperati/Desktop/VALERIO_SPERATI/SORGENTI/ICUB/icub-contrib-common/build/share/ICUBcontrib:/home/valeriosperati/Desktop/VALERIO_SPERATI/SORGENTI/ICUB/icub-main/app
GDMSESSION=lightdm-xsession
SHLVL=1
XDG_SEAT=seat0
HOME=/home/valeriosperati
LANGUAGE=en_GB:en
ICUB_TUT=/home/valeriosperati/Desktop/VALERIO_SPERATI/SORGENTI/ICUB/icub-tutorials/src/build
GAZEBO_PLUGIN_PATH=:/home/valeriosperati/Desktop/VALERIO_SPERATI/SORGENTI/gazebo-yarp-plugins/build/lib
LOGNAME=valeriosperati
XDG_DATA_DIRS=/usr/share/xfce4:/usr/local/share/:/usr/share/:/usr/share
DBUS_SESSION_BUS_ADDRESS=unix:abstract=/tmp/dbus-QEuKyzdHwv,guid=5cef5cdd38498458ea40326b58e20e94
PYCHARM_DIR=/home/valeriosperati/Desktop/VALERIO_SPERATI/SORGENTI/pycharm-community-2016.2.2/bin
XDG_RUNTIME_DIR=/run/user/1000
DISPLAY=:0.0
GLADE_CATALOG_PATH=:
XDG_CURRENT_DESKTOP=XFCE
GAZEBO_YARP=/home/valeriosperati/Desktop/VALERIO_SPERATI/SORGENTI/gazebo-yarp-plugins/build
COLORTERM=xfce4-terminal
XAUTHORITY=/home/valeriosperati/.Xauthority
_=/usr/bin/env

```

claudiofantacci, 2017-04-03:
Nothing seems wrong to me.
It is strange that something is going wrong with the parameters as we are not experiencing this issue on any OS.

The very last thing we should try to do is to completely delete the `build` folder.
Create it again, be sure to be on the proper branch and compile YARP and test `yarpmotorgui` again.

Remember that:
`MASTER branch` -> `yarpmotorgui --name <robot name>`
`DEVEL branch` -> `yarpmotorgui --robot <robot name>`

barbalberto, 2017-04-04:
... mmm ... maybe is it because yarpmotorgui.ini file is not found, so it starts with hardcoded config?
@claudiofantacci is your motorgui finding the `.ini`? If so can you try to run the gui in such a way that the file is not found and see if the behaviour matches with the issue of @valeriosperati?

claudiofantacci, 2017-04-04:
@barbalberto I have also tested that:
```
Claudio@MBP-di-Claudio:~/yarp/build-master/bin (devel) $ ./yarpmotorgui --name icubSim
Warning: QT_DEVICE_PIXEL_RATIO is deprecated. Instead use:
   QT_AUTO_SCREEN_SCALE_FACTOR to enable platform plugin controlled per-screen factors.
   QT_SCREEN_SCALE_FACTORS to set per-screen factors.
   QT_SCALE_FACTOR to set the application global scale factor.
||| configuring
||| default config file specified as yarpmotorgui.ini
||| checking [/Users/Claudio/yarp/build-master/bin/yarpmotorgui.ini] (pwd)
||| checking [/Users/Claudio/Library/Application Support/yarp/config/robots/default] (robot YARP_CONFIG_HOME)
||| checking [/Users/Claudio/Library/Application Support/yarp/robots/default] (robot YARP_DATA_HOME)
||| checking [/Library/Preferences/yarp/robots/default] (robot YARP_CONFIG_DIRS)
||| checking [/etc/yarp/robots/default] (robot YARP_CONFIG_DIRS)
||| checking [/usr/local/share/yarp/robots/default] (robot YARP_DATA_DIRS)
||| checking [/usr/share/yarp/robots/default] (robot YARP_DATA_DIRS)
||| checking [/usr/local/share/yarp/config/path.d] (robot path.d YARP_DATA_DIRS)
||| found /usr/local/share/yarp/config/path.d
||| checking [/usr/local/share/iCub/robots/default] (robot yarp.d)
||| checking [/Users/Claudio/Library/Application Support/yarp/config/yarpmotorgui.ini] (YARP_CONFIG_HOME)
||| checking [/Users/Claudio/Library/Application Support/yarp/yarpmotorgui.ini] (YARP_DATA_HOME)
||| checking [/Library/Preferences/yarp/yarpmotorgui.ini] (YARP_CONFIG_DIRS)
||| checking [/etc/yarp/yarpmotorgui.ini] (YARP_CONFIG_DIRS)
||| checking [/usr/local/share/yarp/yarpmotorgui.ini] (YARP_DATA_DIRS)
||| checking [/usr/share/yarp/yarpmotorgui.ini] (YARP_DATA_DIRS)
||| checking [/usr/local/share/iCub/yarpmotorgui.ini] (yarp.d)
||| did not find yarpmotorgui.ini
yarp: cannot read from yarpmotorgui.ini
[DEBUG]Robot name: icubSim
Setting default parts.
[DEBUG]Appending head
[DEBUG]Appending torso
[DEBUG]Appending left_arm
[DEBUG]Appending right_arm
[DEBUG]Appending left_leg
[DEBUG]Appending right_leg
[INFO]Cancel Button pressed. Closing..
```
yet I get the correct behaviour.

As a matter of the fact, the `--name`/`--robot` parameter overwrites any previously declared parameter under that name.
What it seems to me is that the terminal @valeriosperati is using cannot digest command line parameters.

@valeriosperati, what command line program are you using?
Do you experience this behaviour also with other program?

valeriosperati, 2017-04-04:
I'm using Xfce4-terminal 0.6.3 (bash)
I never experienced any problem with it (in other programs I successfully pass command line arguments).
I tried right now to run the `yarpmotorgui` with xterm, but nothing changes.

I noticed right now a a strange thing: if I launch
`yarpmotorgui --name`
the blank argument is correctly catched

claudiofantacci, 2017-04-05:
@valeriosperati that is even more strange. Since it is able to parse the empty string I do not understand why it is not able to parse what comes next.
Try with:
`yarpmotorgui --name "icubSim"`
or
`yarpmotorgui --name 'icubSim'`

If this is not working either, I am unable to provide any further assistance since we are not able to reproduce this issue on any terminal of any OS.

valeriosperati, 2017-04-05:
...unfortunately it does not work. I think I can try to modify my yarp/src/yarpmotorgui/main.cpp adding some additional message in order to see where command line arguments are lost. I'll let you know. Thanks, Valerio

claudiofantacci, 2017-04-05:
Thanks @valeriosperati, let us know.

valeriosperati, 2017-04-06:
Today I run this test:
I modified the file yarp/src/motorgui/main.cpp adding some cout (look for 'MYSTERY').

```
int main(int argc, char *argv[])
{
//*********************************************
cout<<"\n-----------MYSTERY 1----------\n";    

    std::cout<<"command lines arguments: "<<argc<<"\n";
    std::cout<<"list: \n";
for(int i=0; i<argc; i++)
{
    std::cout<<"["<<i<<"]: "<<argv[i]<<"\n";
}
cout<<"\n-------------------------------\n";    

//********************************************+

    qputenv("QT_DEVICE_PIXEL_RATIO", QByteArray("auto"));
    QApplication a(argc, argv);

    //YARP_REGISTER_DEVICES(icubmod)

    Network yarp;

    if (!yarp.checkNetwork())
    {
        LOG_ERROR("Error initializing yarp network (is yarpserver running?)\n");
        QMessageBox::critical(0,"Error","Error initializing yarp network (is yarpserver running?)");
        return 1;
    }

    ResourceFinder *finder;

    Property p, q;
    finder = new ResourceFinder;

   

    //retrieve information for the list of parts
    finder->setVerbose();
    finder->setDefaultConfigFile("yarpmotorgui.ini");
    finder->setDefault("name", "icub");
    finder->configure(argc,argv);

    //***********************
    cout<<"\n-----------MYSTERY 2----------\n";    

    std::string str=finder->find("name").asString().c_str();
    cout<<"after configure, resource finder finds name: "<<str;
    cout<<"\n------------------------------\n";    

    //************************
```

Running `yarpmotorgui --name icubSim`  the console output is:

```

-----------MYSTERY 1----------
command lines arguments: 3
list: 
[0]: yarpmotorgui
[1]: --name
[2]: icubSim

-------------------------------
||| configuring
||| default config file specified as yarpmotorgui.ini
[... 70 lines cut ...]
||| checking [/home/valeriosperati/Desktop/VALERIO_SPERATI/SORGENTI/ICUB/icub-main/app/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/home/valeriosperati/.config/yarp/yarpmotorgui.ini] (YARP_CONFIG_HOME)
||| checking [/home/valeriosperati/.local/share/yarp/yarpmotorgui.ini] (YARP_DATA_HOME)
||| checking [/etc/xdg/yarp/yarpmotorgui.ini] (YARP_CONFIG_DIRS)
||| checking [/home/valeriosperati/Desktop/VALERIO_SPERATI/SORGENTI/yarp/build/share/yarp/yarpmotorgui.ini] (YARP_DATA_DIRS)
||| checking [/home/valeriosperati/Desktop/VALERIO_SPERATI/SORGENTI/ICUB/icub-main/build/share/iCub/yarpmotorgui.ini] (YARP_DATA_DIRS)
||| checking [/home/valeriosperati/Desktop/VALERIO_SPERATI/SORGENTI/ICUB/icub-contrib-common/build/share/ICUBcontrib/yarpmotorgui.ini] (YARP_DATA_DIRS)
||| checking [/home/valeriosperati/Desktop/VALERIO_SPERATI/SORGENTI/ICUB/icub-main/app/yarpmotorgui.ini] (YARP_DATA_DIRS)
||| did not find yarpmotorgui.ini
yarp: cannot read from yarpmotorgui.ini

-----------MYSTERY 2----------
after configure, resource finder finds name: icubSim
------------------------------
[DEBUG]Robot name: icub
Setting default parts.
[DEBUG]Appending head
[DEBUG]Appending torso
[DEBUG]Appending left_arm
[DEBUG]Appending right_arm
[DEBUG]Appending left_leg
[DEBUG]Appending right_leg
```

This really puzzles me. String 'foo' is correctly parsed by the finder, string 'name' is not parsed.

claudiofantacci, 2017-04-06:
Your problem seems related to the resource finder then.

Try commenting out `finder->setDefault("name", "icub");` and do the test again to see what happens.

valeriosperati, 2017-07-24:
The problem disappeared on last git update (both icub and yarp)!

closed 2017-07-24 by valeriosperati