robotology/yarp #1156: 'yarp server --ros' on a windows machine leads to segfault

randaz81, 2017-03-30:
segfault @ Contact RosNameSpace::rosify(const Contact& contact)  rosNameSpace.cpp
if hostname variable is fixed, the segfault does not happens but ros communication simplly does not work

aerydna, 2018-05-29:
cannot reproduce the issue. we suspect it to be already fixed, if not, reopen.

randaz81, 2018-05-29:
fixed by what?

randaz81, 2018-05-29:
Since the issue is probably a missing communication layer in WSL, let's keep this open until Microsoft releases the next patch and let's try again.

drdanz, 2018-05-29:
@randaz81 can you explain how to reproduce this? We were not able to reproduce this with a MSVC build, but perhaps this happens on WSL only?

traversaro, 2018-05-29:
If there is some system call problem in WSL, you can check/report it in https://github.com/Microsoft/WSL/issues .

drdanz, 2019-04-11:
I can no longer reproduce this.
I suspect this was fixed when I changed `NI_MAXHOST` to `HOST_NAME_MAX`.
Anyway, If this is a WSL error, it does not belong here. Closing

closed 2019-04-11 by drdanz