robotology/yarp #1165: Yarprobotinterface segfault after firmware update

pattacini, 2017-04-04:
@dcam0050 commented on [Tue Apr 04 2017](https://github.com/robotology/icub-support/issues/438)

### Description of the failure

I have updated `yarp`, `icub-main`, `icub-firmware-shared` and `icub-firmware-build` to their latest version on pc104. Then I updated the firmware for all ETH and all the CAN boards,  

Then after I power cycled the motors and pc104 I ran yarprobotinterface and I got a Segmentation fault error late in the booting process.

### Detailed conditions and logs of the failure
I've attached here the log for yarprobotinterface.

[yarprobotinterface_log.txt](https://github.com/robotology/icub-support/files/893780/yarprobotinterface_log.txt)

I've tried compiling yarp with libmath using eigen3 and libmath using gsl to check if it makes a difference but it still fails in the same spot.

Any help would be greatly appreciated! Thanks

---

@pattacini commented on [Tue Apr 04 2017](https://github.com/robotology/icub-support/issues/438#issuecomment-291553934)

Hi @dcam0050, as per our [guidelines](https://github.com/robotology/icub-support/blob/master/.github/CONTRIBUTING.md), this repo is only for tracking requests for robot maintenance. Thus, I'll move this issue to yarp.

dcam0050, 2017-04-04:
This was fixed by updating pc104 image and re-compiling yarp and icub-main. Also required compiling embobjMAIS and embObjStrain which are not included in the compilation for pc104 wiki page.

Nicogene, 2017-05-05:
Which is the status of this issue?? It can be closed?

pattacini, 2017-05-05:
I think so...

closed 2017-05-05 by pattacini