robotology/yarp #1167: yarpserver becomes unresponsive when connecting and disconnecting ports

rlober, 2017-04-05:
Hello all,

**YARP version 2.3.69.7 on `devel`**

I am working on a python application where I have a a buffered port `receiver` which tries to connect to a streaming port, `/writer`, once it opens. Once connected, `/receiver`  listens until `/writer` stops sending and is closed. `/receiver` then goes back to waiting for a connection to `/writer` and this process is repeated. After the first disconnection, when the `/receiver` starts trying to connect again with `/writer` `yarpserver` becomes unresponsive. It doesn't crash but nobody can connect to it. 

I have written two python scripts which reproduce this error/behavior.

## `writer.py`
```py
import time
import yarp

if __name__ == "__main__":
    yp = yarp.Network()
    yp.init()

    port = yarp.Port()
    btl = yarp.Bottle()
    for i in range(5):
        print("Starting up writer port")
        port.open("/writer")

        for j in range(200):
            btl.clear()
            btl.addInt(j)
            port.write(btl)
            time.sleep(0.01)

        print("Closing writer port")
        port.close()
        time.sleep(5)
```

## `receiver.py`
```py
import yarp
import time

if __name__ == "__main__":
    yp = yarp.Network()
    yp.init()

    port_in = yarp.BufferedPortBottle()
    port_in.open("/receiver")
    n_connects = 0
    while n_connects<5:
        if yp.connect("/writer", "/receiver"):
            print("Connected")
            while yp.isConnected("/writer", "/receiver"):
                btl = port_in.read(False)
                if btl is not None:
                    print(btl.get(0).asInt())

            print("Disconnected.")
            n_connects += 1

    port_in.close()
```

To launch the example, open 3 terminals and run in the following order:

### Terminal 1
```
yarpserver
```

### Terminal 2
```
python3 receiver.py
```

### Terminal 3
```
python3 writer.py
```

You will get the following output:

**`writer.py` ouput**
```
Starting up writer port
yarp: Port /writer active at tcp://10.255.35.111:10003/
yarp: Sending output from /writer to /receiver using tcp
Closing writer port
yarp: Removing output from /writer to /receiver
Starting up writer port
yarp: Cannot connect to port /ryan
yarp: Port /writer failed to activate (invalid address)
```

**`receiver.py` ouput**
```
yarp: Port /receiver active at tcp://10.255.35.111:10002/
yarp: Receiving input from /writer to /receiver using tcp
Connected
1
2
3
...
198
199
yarp: Removing input from /writer to /receiver
Disconnected.
yarp: No connection to nameserver
yarp: *** try running: yarp detect ***
```

Any ideas what's happening?

@traversaro @francesco-romano @drdanz

barbalberto, 2017-04-06:
I just tried using python 2.7 and it works fine for me, using your scripts. Which python vesion do you use?

rlober, 2017-04-06:
3.5.2

rlober, 2017-04-06:
So after some investigation with @drdanz we found the following problem: 

When the `yarpserver` is started and the `receiver` is launched, after a few seconds it looses connection with the `yarpserver`. We close the `receiver` then `yarp clean` then rerun the `receiver` and we do not see the same issue. One of the problems is the fact that the `receiver` is querying `yarpserver` too quickly. Adding a delay of 10ms between `yarp.connect(...)` calls, fixes this failure mode. 

```py
import yarp
import time

if __name__ == "__main__":
    yp = yarp.Network()
    yp.init()

    port_in = yarp.BufferedPortBottle()
    port_in.open("/receiver")
    n_connects = 0
    while n_connects<5:
        if yp.connect("/writer", "/receiver"):
            print("Connected")
            while yp.isConnected("/writer", "/receiver"):
                btl = port_in.read(False)
                if btl is not None:
                    print(btl.get(0).asInt())

            print("Disconnected.")
            n_connects += 1
        time.sleep(0.01)
    port_in.close()
```
Our current work around is to use `Persistent` connections and look at the number of connected ports using: `port_in.getInputCount() > 0`

I will let @drdanz and @barbalberto decide what to do with this issue, since the root problem still exists but may not represent a common use case.

rlober, 2017-04-06:
Here is the *fixed* `receiver` code:

### `receiver.py`
```py
import yarp
import time

if __name__ == "__main__":
    yp = yarp.Network()
    yp.init()

    port_in = yarp.BufferedPortBottle()
    port_in.open("/receiver")
    style = yarp.ContactStyle()
    style.persistent = True
    yp.connect("/writer", "/receiver", style)
    n_connects = 0
    port_was_connected = False
    while n_connects < 5:
        while port_in.getInputCount() > 0:
            if not port_was_connected:
                print("Connected")
                port_was_connected = True
            btl = port_in.read(False)
            if btl is not None:
                print(btl.get(0).asInt())

        if port_was_connected:
            print("Disconnected.")
            n_connects += 1
            port_was_connected = False

    yp.disconnect("/writer", "/receiver", style)
    port_in.close()

```

Note that the connection needs to be removed manually using the line `yp.disconnect("/writer", "/receiver", style)`, or else the persistent connection will remain even after the scripts finish.

drdanz, 2017-05-16:
@rlober I think we close this for now... Please re-open if you think there is something that we could fix in yarp...

closed 2017-05-16 by drdanz