robotology/yarp #1168: yarprobotinterface -> seg fault

pattacini, 2017-04-06:
@davidetome commented on [Thu Apr 06 2017](https://github.com/robotology/icub-main/issues/436)

Every time CTRL+C is sent yarprobotinterface returns "Segmentation fault (core dumped)" and then you must reset the boards to run it again

---

@pattacini commented on [Thu Apr 06 2017](https://github.com/robotology/icub-main/issues/436#issuecomment-292102624)

I'm afraid then that the problem involves also the FW.

cc @robotology/firmware-developers

pattacini, 2017-04-06:
See #745

marcoaccame, 2017-06-01:
The segfault should not happen, but if it happens it is wrong that one has to reset the boards before to relaunch yarprobotinterface. There is something which must be fixed.

For this reason, I had a look at that. See issue [ref] for details.

marcoaccame, 2017-06-01:
See the bugfix in https://github.com/robotology/icub-main/issues/452. Now after a crash of yarprobotinterface the ETH boards do no need to be restarted again.

Dear @davidetome, now the CTRL-C causes less problems.

Dear @pattacini and @drdanz it remains only the crash problem of yarprobotinterface which does not depends from firmware and I think it is of complex solution (at least because it is a very long standing issue).

drdanz, 2017-06-07:
@marcoaccame @pattacini you fixed something on the cartesian controller a few days ago... is the crash fixed now? Can we close this?

pattacini, 2017-06-07:
the crash is fixed yep 👍

closed 2017-06-07 by pattacini