robotology/yarp #1170: Problem overwriting group param from terminal

jeljaik, 2017-04-10:
Hello guys, 

I'm trying to overwrite the value of a group parameter from console and what I'm getting is just an additional param/value pair added to the corresponding group. 

This is how I'm passing the new value from terminal, as instructed [here](http://wiki.icub.org/codyco/dox/html/yarp_config_files.html#yarp_config_command).

``` 
walking-client --MIQP_CONTROLLER_PARAMS::N 20
```
Then after configuring the RF, if I do `rf.toString()` I get the following:
```
rf (period 10) (from walking-client.ini) (robot icubGazeboSim) (MIQP_CONTROLLER_PARAMS (N 20)  (dCoMxRef 0.0) (dCoMyRef 0.0) (N 15)
```
You can see how the list for `MIQP_CONTROLLER_PARAMS` now contains `(N 15)` and `(N 20)` instead of overwriting the value in my config file.

I'm using `yarp` 2.3.66.

Thanks

Nicogene, 2017-05-05:
Hi @jeljaik,

this behavior is reproducible also using the ini files, e.g. `yarpdev --from test_grabber.ini` where `test_grabber.ini` is:
```
device test_grabber
[MIQP_CONTROLLER_PARAMS]
N 15
N 20
```
you will have a `Property` like this:
```
(MIQP_CONTROLLER_PARAMS (N 15) (N 20)) (device test_grabber)
```
The problem is that the groups are `Bottle` and `Bottle` have not any uniqueness constraint, so the other `N` will be add to the bottle without overwriting the old value.
`Bottle` inherits from `Searchable` `find*` methods that use a key-value mechanism to find an element that in this case has NOT a "map" structure.
The problem is quite articulated, the ways to solve could be two:

- Make the groups `Property`
- Add the uniqueness constraint to `Bottle`

I don't know which is more feasible, I'm sure that both options are painful :smile:

traversaro, 2017-05-05:
I think that making the groups `Property` is the correct way to go, the `Bottle` class has been used for too many year to change its semantics/behavior.

lornat75, 2017-05-10:
I agree!